rdaly525/coreir #26: Update Python c bindings to correctly find shared library files

rdaly525, 2017-04-21:
Use branch 'libreorg' in order to:
1) Change python bindings  to expect the shared libraries to come from bin/
1) Have an easy way of loading coreir-libs into python.

The load library call will always have a C++ function signature:
Namespace* CoreIRLoadLibrary_<libname>(Context* c);

For example for stdlib:
Namespace* CoreIRLoadLibrary_stdlib(Context* c);

leonardt, 2017-04-24:
Tracked by https://github.com/rdaly525/coreir/pull/29

leonardt, 2017-04-25:
I imagine the Python interface to look something like 
```python
ctx = Context()
stdlib = ctx.load_library("stdlib")
```
To do this, it could dispatch on the argument to `load_library` and call the correct function `coreir_stdlib.CoreIRLoadLibrary_stdlib(ctx)` but either:
* The load library call is defines an `extern C` function and uses the `coreir-c` opaque types.
* The library registers itself with the C-api which holds a map between strings to library values and provides a C-api function `coreir_lib.LoadLibrary(ctx, name)` which handles converting the `Namespace*` and `Context*` between cpp and opaque types.

rdaly525, 2017-04-25:
@leonardt In regards to #2, how does the library register itself with the C-api? I think it would still necessarily involve writing an extern C function per library. Unless I am misunderstanding something...

That being said, it would be cool to be able to register library functions which would mean no need for a specific function name standard.

rdaly525, 2017-04-27:
@leonardt I just pushed to the clibfix branch (off of libreorg) the addition of the C stdlib api. There are three TODOs (one in lib/libs/stdlib.cpp, two in include/coreir-libs/stdlib.h)

It would be awesome if you could make macros (defined in include/coreir.h) that would macrofy each of the 3 TODOs.

Also now that we have the C api for the libs, you can add them to the python bindings

leonardt, 2017-04-27:
README updated [link](https://github.com/rdaly525/coreir/tree/clibfix#libraries).
@rdaly525 Should I submit a PR for this branch or are there other fixes pending? I'll wait to close this issue until we merge the branch.

closed 2017-04-28 by leonardt