rdaly525/coreir #27: Generator Params vs. TypeGen Params

THofstee, 2017-04-22:
Currently, the Generator constructor requires that the Params provided to the Generator match the Params provided to the TypeGen. There are cases where you would like the Generator to take additional parameters over the TypeGen. For example, consider a line buffer.

The line buffer needs to know at least 5 things: the width and height of the taps you want, the width of the image, the height of the image, and the bits per pixel. On the other hand, the type of the line buffer is very easily represented, it is just a single pixel in, and width*height pixels out. The TypeGen does not require any knowledge about the width and height of the image.

Instead, I propose that the Parameters provided to the Generator be a superset of the Parameters provided to the TypeGen.

THofstee, 2017-04-22:
An argument against would be we would need to know the image width in order to know the delay of a module, but I'm going to disagree with that by saying the delay of a module should be a property of a module, not a property of its interface.

I'm open to hear what anyone else thinks.

rdaly525, 2017-04-26:
I think this is good and allows typegens to be able to be used by a larger set of generators.

closed 2017-05-02 by rdaly525