rdaly525/coreir #33: API Problems and Requests

rdaly525, 2017-04-26:
This is a general issue for discussing problems with the coreIR API

Please comment on this github issue with API problems labeled with C++, C, or Python

Does a function cause a segfault? (If so please file a separate issue)
Does a function cause coreIR to falsely error?
Does a function produce unhelpful error messages?
Is the name of the function confusing?
Is there some basic function you want and cannot do?

THofstee, 2017-04-28:
`libs/stdlib.hpp` has `using namespace CoreIR;` as one of the lines in the file. This seems like a potentially bad idea because now any file that #includes stdlib will now also be using the CoreIR namespace.

rdaly525, 2017-04-28:
Totally agree. This should be added to the issue #32

THofstee, 2017-04-28:
Can confirm that passing in nullptr as the Type* to runGenerator causes a segfault.

jeffsetter, 2017-05-02:
The formatting for 2+ dimensional arrays seems a bit cumbersome. Perhaps some syntactic sugar to make specifically arrays easier to write?

      {"out",context->Array(3, context->Array(3,
                                              context->Array(bitwidth,context->BitOut())  )) }

rdaly525, 2017-05-02:
@jeffsetter What is a proposed syntax?

jeffsetter, 2017-05-02:
I was thinking that C's array syntax is fairly standard, so something like
context->BitOut()[bitwidth][3][3]

I'm not sure if this would be annoying to implement, but it looks a little cleaner to me.

rdaly525, 2017-05-02:
@jeffsetter I have thought about doing that, the problem is that Type is a pointer. And you would have to dereference it in order to use the [] operator. 

So it would have to look like:
Type* stencil = *(*(*(context->BitOut())[bitwidth])[3])[3]

Which I would argue is uglier.

I could do:
Type* stencil = context->BitOut()->Arr(bitwidth)->Arr(3)->Arr(3)
which is a bit better.

rdaly525, 2017-05-02:
Added in #48

THofstee, 2017-05-10:
I believe coreir.h and coreir-c.h should replace `extern COREModuleDef* COREModuleGetDefs(COREModule* m);` with `extern COREModuleDef* COREModuleGetDef(COREModule* m);`. There is no such function named `COREModuleGetDefs`.

cdonovick, 2017-05-16:
There is currently no way to determine the bitwidth of a connection in python beyond inspecting the selectpath of both end points and searching for `'data'` or `'bit'` and making the assumption that:
`*.data.out -> *.data.in[*]` implies 16 bit connection and
`*.bit.out    -> *.bit.in[*]` implies 1 bit connection

It would be preferable if connections had their width.

rdaly525, 2017-05-16:
@caleb, opened new issue #71

THofstee, 2017-05-18:
C interface segfaults when calling `CORELoadModule` on `_add4Gen.json` from unit tests.

closed 2017-10-24 by rdaly525