rdaly525/coreir #34: Python API should never cause CoreIR to die

rdaly525, 2017-04-26:
Update C and Python bindings such that you cannot get coreIR to die using the Python API

leonardt, 2017-05-15:
The recent bug in dview makes me motivated to tackle this error handling.

@rdaly525 any opinion on whether we should have the C-API functions return error codes or use an error flag passed as a parameter?

rdaly525, 2017-05-16:
I like the llvm approach. The majority of functions don't have errors.

Have "has", "get", and "set" functions.
Basically all getter functions return null if it can't get the value. To do error checking here, you should always call the has function first. 

In terms of thing she like validate/typechecking that should definitely pass in a bool* or return a bool. But the number of these types of functions should be relatively small.

closed 2017-10-24 by rdaly525