rdaly525/coreir #35: Try SWIG for generating API wrapper

leonardt, 2017-04-26:
http://www.swig.org/

Could allow us to automatically generate Python and Lua bindings

THofstee, 2017-05-03:
Here's another option: https://github.com/google/clif

rdaly525, 2017-05-03:
I would be a little worried because this goes from C++ to python. This might not be good if we want to interface with Lua too.

The ideal thing would be to write a single C header file (coreir.h), and then automatically generate the Lua and Python bindings.

leonardt, 2017-06-20:
Should consider the experience report from the Halide devs found at https://github.com/halide/Halide/tree/master/python_bindings (excerpted below).

## Why did we move from SWIG to Boost.Python ? ##

Problems with (old) SWIG bindings:

- P1) SWIG cannot parse C++15, and it will always lag
- P2) Python code is not compiled/tested together with C++ codebase
- P3) Albeit SWIG is ment to provide "automagic" conversion, current code base has explicit code for every class (and class member) of interest
- P4) Old swig bindings `__init__.py` file has a lot of "python magic" to get things running, code is spread around multiple pieces (`py_util.cpp`, `cHalide.i`, and `__init__.py`)
- P5) Old swig python bindings did not work anymore (swig issues, compilation issues)

The boost python approach helps because:
- a) P1 is solved out of the box, since no C++ parsing is involved
- b) P2 is mitigated since python bindings are also in C++. Any miss-match with generate a compilation error.
- c) P3, P4 and P5 are expected to be handled by having less spaghetti and aiming for tighter integration with rest of code-base (point b)

rdaly525, 2017-06-20:
This is all complaints about C++ code. It would be useful to figure out if these problems also exist for C code