rdaly525/coreir #40: passes.hpp file extension

jeffsetter, 2017-05-01:
All other header files in the include directory use .h while include/coreir-pass/passes.hpp uses a .hpp extension. I propose to switch this file to passes.h for consistency.

THofstee, 2017-05-02:
Because there are a mix of c and c++ files in this repo now, maybe it would be better to name any C headers with .h and any C++ headers with .hpp?

rdaly525, 2017-05-02:
I think in general that should be the case, but for anything in include/* we should name it as .h so that we can include them in the c api.

In the Readme, we have a well defined way of adding libraries so that we can use it in both the C api and the c++ api. 

Passes should probably do something similar.

rdaly525, 2017-05-14:
This should be fixed by pull request #67

closed 2017-05-14 by rdaly525