rdaly525/coreir #71: Ability to get type and bit width of connection

rdaly525, 2017-05-16:
Need to add this in C++, C, and python API

Option 1:
implement Type.getRawWidth()

Option 2:
Port over the isa, and cast functions, and ArrayType.getLen()

rdaly525, 2017-05-17:
I think the correct solution is to use issue #73 solution. Basically just have a function COREGetLen(COREType* arraytype)

rdaly525, 2017-06-01:
@leonardt Can you add the Python api for Type.get_len()? You should throw an error if the type is not an array type. @cdonovick wanted this so he does not have to hardcode things.

THofstee, 2017-06-01:
I actually hacked this in to my branch. I just did the following:

```
COREType* COREWireableGetType(COREWireable* w) {
    return rcast<COREType*>(rcast<Wireable*>(w)->getType());
}
```

rdaly525, 2017-06-06:
Also implemented getSize on types.

closed 2017-06-06 by rdaly525