rdaly525/coreir #73: C API updates (class inheritance)

rdaly525, 2017-05-17:
I think it would be good to make consistent and simplify the C API. It is annoying to port over class inheritance structures. And I think I have been a bit ad hoc about it.

In general if there is an inheritance hierarchy, only define the Base class type for the C API. And have all the functions in all of the class hierarchy take in that one Base class type and internally cast it.
For example, change COREInstance to COREWireable.
In addition, have enumeration of all the 'kind's. COREWireableGetKind() This will return an enumeration of either a select, an instance, or an interface. This allows one to determine the type. 

Concrete tasks

- [ ] Remove CORESelect and COREInstance, and replace with COREWireable. 
- [ ] Add COREWireableGetKind() and an enumeration for the wireable kinds
- [ ] Remove CORE*Type and replace with COREType
- [ ] Add CORETypeGetKind() and an enumeration for the type kinds
- [ ] Remove CORE*Arg and replace with COREArg
- [ ] Add COREArgGetKind(). The enumeration should just be COREParam

All the functions should use cast<> or dyn_cast<> to cast to the correct c++ type. Note: cast<> will assert on failure. dyn_cast returns null on failure.

rdaly525, 2017-05-24:
@leonardt I am working in branch capi. The C api is working, and the python bindings need updating.
Summary of changes can be found in the commit message. Let me know if you have any questions.

rdaly525, 2017-05-24:
@THofstee These changes might affect you too. I am in the process of refactoring/simplifying the C api.

rdaly525, 2017-05-24:
Also I annoyingly renamed the COREInt2Arg and COREArg2Int functions to be more consistent with the C++ naming conventions.

leonardt, 2017-05-24:
I'll probably want to wait for https://github.com/rdaly525/coreir/pull/76 to be merged before I make more changes to the Python bindings. Or at least rebase `capi` off of it so I don't have to deal with the conflict later.
@cdonovick would merging https://github.com/rdaly525/coreir/pull/76 break your code? Or are you working off of master

THofstee, 2017-05-24:
I'll take a look. In the c api in the dev branch there are a number of misc bugs - multiple cases of using pointers to strings that have been deleted on the stack, and a few cases of nondeterministic behavior (sometimes get path length of 2 for src, or 3, or a type casting error), among others that I've been fixing as I come along.

THofstee, 2017-05-25:
At a quick glance it looks like most of the bugs are still present in the new version. Regardless I'm going to rebase my branch off the new code.

rdaly525, 2017-05-25:
@THofstee If you can create a test that causes bad behavior, definitely file a bug on me so I can take a look.

THofstee, 2017-05-25:
It's not so much test cases, but actual use of the API. It seems that a few of the ones that were bugs before have been changed in the underlying structure so they are non issues, but for example:

`const char* COREGetInstRefName(COREWireable* iref) {
	string name = rcast<Instance*>(iref)->getModuleRef()->getName();
	return name.c_str();
  }`

should be more like:
`const char* COREGetInstRefName(COREWireable* iref) {
	string name = rcast<Instance*>(iref)->getModuleRef()->getName();
	size_t size = name.length()+1;
	char* res = (char*)malloc(size);
	memcpy(res, name.c_str(), size);
	return res;
  }`
If you wanted to do this like you've done for the others, it needs a new function in context to register the allocated memory to be deleted at context deletion.

because in the first case, you're creating a `const char*` to the string object, which is constructed on the stack, then deleted at function exit so when you try to access the string from something using the API you get garbage values.

In the C header file you also have a few issues, notably:
`extern COREModuleDef* COREModuleGetDef(COREModule* m);` is commented out
`extern const char** COREDirectedConnectionGetSrc(COREDirectedConnection* directed_connection);
extern const char** COREDirectedConnectionGetSnk(COREDirectedConnection* directed_connection);`
should be
`extern const char** COREDirectedConnectionGetSrc(COREDirectedConnection* directed_connection, int* path_len);
extern const char** COREDirectedConnectionGetSnk(COREDirectedConnection* directed_connection, int* path_len);` to match the function signature in the C implementation.

Lastly, it seems like making a `DirectedModule` from a `Module`, then calling `DirectedModuleGetOutputs`, followed by `DirectedConnectionGetSnk/Src` results in getting the source/sink of a module that is an instance within the module we called it with, but not that module itself. Not sure if this last one is my fault or a bug though.

cdonovick, 2017-05-25:
I working off coreir master currently

rdaly525, 2017-05-25:
is the char* created on the stack? I thought that c_str() just pointed to the internal char* within std::string. I am confused why that is not the case.

THofstee, 2017-05-25:
It's valid as long as the string is valid. The string is created in the function, then destroyed at exit. If it was a string reference it would be better, but this still didn't work in my case, likely due to the getName() function doing something similar if I had to guess.

THofstee, 2017-05-25:
Forgot to add: Valid assuming no modifications to the string have been made in the meanwhile.

THofstee, 2017-05-25:
Yeah it looks like getName() returns a string. If that was changed to return a string& and you are fine with users having direct access to that memory (which I think is the right choice here, their fault if they invalidate it) then changing this function to just return the c_str() (either without the temp string allocation, or just calling c_str() after getName() and returning that call directly) then I think this would work as originally intended.

rdaly525, 2017-05-25:
Ah I see the issue now. Let me check how LLVM deals with this kind of thing

rdaly525, 2017-05-25:
Yeah, I agree that I should just return a reference to the string.

THofstee, 2017-05-25:
Ok the issue I had with DirectedModule seemingly returning unrelated inputs/outputs was my fault -- Lua indexing starting from 1 threw me off.

rdaly525, 2017-05-31:
Fixed by #83

closed 2017-05-31 by rdaly525