rdaly525/coreir #95: Support getting the width of a dconnection

leonardt, 2017-06-13:
Caleb needs this for PNR.
Syntax proposals: `len(connection)` or `connection.width`
Should be simple for a normal connection (just get the width of one of the wireables)

For directed connections (which Caleb is using) the current way to do it is a bit clunky (need to follow the select path to get a wireable which you can query for the type and then width). This is because connection.source returns a select path instead of a wireable. @rdaly525 Could directedconnection.{source,sink} return Wireables instead?

rdaly525, 2017-06-13:
If this is urgent, I would say just implement the python len() function for connection first by abstracting away the clunkyness to the python bindings. 

There is nothing wrong with getsrc/getSnk returning a Wireable instead, although it is a little bit clunky if you just want to traverse the directed graph.

rdaly525, 2017-06-13:
I would maybe have a connection.size for the raw bit width in a similar way for type

leonardt, 2017-06-13:
I'll do `connection.size` for now to be consistent with the type API.

closed 2017-07-19 by leonardt