rdaly525/coreir #100: Migrate C/Python API to use iterator pattern

leonardt, 2017-06-19:
Iterator API
* `CORE{type}Get{member}Iter` Returns an iterator over `member`
* `COREIterFirst` Get the first item from an iterator
* `COREIterNext` Get the next item from an iterator
* `COREIterLast` Get the last item from an iterator
This would allow us to reuse COREIter{First,Next,Last} for all *iterable* members throughout the API. That is, each class with an iterable member will just need to implement `CORE{type}Get{member}`.
An alternative approach would be to have explicit functions for each class/member pair:
* `CORE{type}{member}GetFirst`
* `CORE{type}{member}GetLast`
* `CORE{type}{member}GetNext`

Functions that need to be updated
- [ ]  `module_def.instances`
- [ ]  TODO: List out the rest

@rdaly525 Any feedback on this proposed API?

rdaly525, 2017-06-20:
I think the second would be best. I do not see how we could do the first without having void* as the parameter types.

rdaly525, 2017-10-24:
Potentially have a generic iterator pattern