rdaly525/coreir #116: Have make install actually install into the system

leonardt, 2017-07-10:
I propose we separate the notion of building and installing coreir.
* `make` builds the coreir shared libraries into a build directory (basically what `make install` does now)
* `make install` copies the build into the system path (i.e. `/usr/lib/` and `/usr/include`)

This moves us towards a `make install` for the entire CGRAFlow (that is, each internal dependency needs a notion of `make install` that builds and installs their software into the system).

rdaly525, 2017-07-10:
I would be okay with separating out the python bindings into a separate repo. It is essentially a separate project on its own that you are maintaining. Kind of like llvmlite. 

I also think separating out the make and make install is a good idea as we would have to do that anyways for a real release of coreir. You probably know better than me how other projects do that.

We should probably create a document alongside with this change to describe the install/use process for new users and all the possible make commands.

rdaly525, 2017-09-09:
@leonardt let's make this a high priority. This should solve the OS X issues that pat has been experiencing

rdaly525, 2017-09-17:
Fixed

closed 2017-09-17 by rdaly525