rdaly525/coreir #133: Create a script to generate all the firrtl primitive op wrappers

rdaly525, 2017-07-28:
This will look similar to the script we have for generating verilog modules.
I think this would be nice so that we could leverage it in the firrtl backends for both magma and coreir. 
What do you think?

leonardt, 2017-07-28:
Sounds reasonable. 

One issue is that my firrtl backend for magma emits a wrapped module for each input width (i.e. Add8 where the input types are UInt<8>). It does this on demand depending on what is used the in the magma circuit being compiled.

I just read in the spec that you can leave the width of the IO types to a module unspecified.
> Alternatively, if the bit width is omitted, it will be automatically inferred by FIRRTL’s width inferencer, as detailed in section 9.

This sounds like a better strategy. Basically, define a generic "Module" version of each primitive with the widths unspecified. I think we still need to define different wrappers for different IO types (i.e. an unsigned add vs a signed add).

leonardt, 2017-07-28:
Also, because it doesn't hurt to ask :)
https://github.com/freechipsproject/firrtl/issues/632

leonardt, 2017-07-28:
From Adam:
> The problem with simply added the inst syntax is that FIRRTL primops are polymorphic in their input/output types. Thus, at declaration of inst my_adder of add, we don't know the input or output types of my_adder.

So looks generating wrapped modules with the inline annotation is our best bet.

rdaly525, 2017-07-28:
Yeah that does make sense. The only other way to do that would be if they could parameterize instances (which I do not think they could do)
inst my_adder_signed of add(SInt)

rdaly525, 2017-07-28:
But yeah, I do agree that it is preferable to use wrappers that are generic over the width. We might as well utilize that feature. I also definitely agree that we will need separate ops for signed and unsigned.
For example something like

`Module firrtl_slt : 
  in : UInt[2]
  out UInt<1>
  in0 <= toSInt(in[0])
  in1 <= toSInt(in[1])
  out <= lt(in0,in1)`

rdaly525, 2017-08-07:
@leonardt, Are you going to be able to do this before the meeting next Monday? Also related, do you know if it is possible to have separate files for the same firrtl circuit?

leonardt, 2017-08-07:
@rdaly525 sure, I can have it done by Wednesday (busying prepping for AHA presentation tomorrow)

rdaly525, 2017-08-08:
Awesome thanks!

leonardt, 2017-08-08:
@rdaly525 I've started on this, but it looks like we're not going to be able to use this approach for parameterized primitives like `shr(e, n)` where `n` is a parameter (static shift). I can't generate a module for each parameter, so it looks like you'll need to either generate these modules on demand (when you know which specific parameter you need) or special case the code generation to handle firrtl primitives directly (that's what I did in magma).

rdaly525, 2017-08-09:
Is that the only case where this is an issue? I could imagine a workaround to be just treat it as a dynamic shift and let firrtl compiler change it to a static shift.

leonardt, 2017-08-09:
ops with parameters are
* `pad`
* `shl`, `shr`
* `bits` (bit extraction, selecting a range of bits)
* `head`, `tail` (truncate `n` bits off the beginning/end)

rdaly525, 2017-10-24:
This is unneeded now

closed 2017-10-24 by rdaly525