rdaly525/coreir #166: Python bindings are broken

rdaly525, 2017-08-21:
When I run the bindings on my mac, I get:

```
tests/bindings/python/test_wireable.py:1: in <module>
    import coreir
bindings/python/coreir/__init__.py:5: in <module>
    from coreir.lib import load_shared_lib, libcoreir_c
bindings/python/coreir/lib.py:19: in <module>
    libcoreir_c = load_shared_lib("c")
bindings/python/coreir/lib.py:17: in load_shared_lib
    return cdll.LoadLibrary('{}/../../../lib/libcoreir-{}.{}'.format(dir_path, suffix, shared_lib_ext))
/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ctypes/__init__.py:440: in LoadLibrary
    return self._dlltype(name)
/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ctypes/__init__.py:362: in __init__
    self._handle = _dlopen(self._name, mode)
E   OSError: dlopen(/Users/rdaly/coreir/bindings/python/coreir/../../../lib/libcoreir-c.dylib, 6): Library not loaded: @rpath/libcoreir.dylib
E     Referenced from: /Users/rdaly/coreir/lib/libcoreir-c.dylib
E     Reason: image not found
```

This is the case for trying to load any library which references libcoreir.dylib. This is NOT broken when I run a C++ test which uses the same library.

leonardt, 2017-08-21:
@rdaly525 I believe you've resolved this or found a workaround, can you post it here? @phanrahan was having problems getting setup with coreir and the error looked similar.

I have a hunch it has to do with relative paths. If so, we can have the library lookup use the coreir environment variables to find the libraries until we have a set system installation path.

rdaly525, 2017-08-22:
Well I basically cannot reproduce it on my system anymore. (I was screwing up the DYLD_LIBRARY_PATH).

if DYLD_LIBRARY_PATH is set correctly we should not see this error.

leonardt, 2017-08-22:
Okay, I will close this issue for now

closed 2017-08-22 by leonardt