rdaly525/coreir #197: After running flatten pass on coreir json file the coreir to verilog pass says it has not been flattened?

dillonhuff, 2017-09-11:
I'm trying to generate verilog from coreir json to feed to verilator.

The json file I am experimenting with was generated from a circuit adding 4 32 bit numbers. The original file is:

```json
{
  "namespaces": {
    "global": {
      "modules": {
        "Add4": {
          "connections": [
            [
              "self.in.3",
              "add01.in1"
            ],
            [
              "self.in.2",
              "add01.in0"
            ],
            [
              "self.in.1",
              "add00.in1"
            ],
            [
              "self.in.0",
              "add00.in0"
            ],
            [
              "self.out",
              "add1.out"
            ],
            [
              "add1.in1",
              "add01.out"
            ],
            [
              "add1.in0",
              "add00.out"
            ]
          ],
          "instances": {
            "add00": {
              "genargs": {
                "width": 32
              },
              "genref": "coreir.add"
            },
            "add01": {
              "genargs": {
                "width": 32
              },
              "genref": "coreir.add"
            },
            "add1": {
              "genargs": {
                "width": 32
              },
              "genref": "coreir.add"
            }
          },
          "type": [
            "Record",
            {
              "in": [
                "Array",
                4,
                [
                  "Array",
                  32,
                  "BitIn"
                ]
              ],
              "out": [
                "Array",
                32,
                "Bit"
              ]
            }
          ]
        }
      }
    }
  }
}
```

When I tried to run the verilog pass directly on this file I got the error:
```bash
bash-3.2$ ./bin/coreir -i ./tests/simulator/add4.json --load_passes ./lib/libcoreir.dylib -p verilog -o add4.v
ERROR: {global.Add4}.in is not flattened!
  Type is: BitIn[32][4]

Assertion failed: (isBitOrArrOfBits(rmap.second)), function runOnInstanceGraphNode, file verifyflattenedtypes.cpp, line 37.
Abort trap: 6

```

As an attempted fix I tried flattening the file first and then running the verilog pass on the flattened json like so:

```bash
bash-3.2$ ./bin/coreir -i ./tests/simulator/add4.json --load_passes ./lib/libcoreir.dylib -p flatten -o add4_flat.json

Modified?: No
bash-3.2$ ./bin/coreir -i add4_flat.json --load_passes ./lib/libcoreir.dylib -p verilog -o add4.v
ERROR: {global.Add4}.in is not flattened!
  Type is: BitIn[32][4]

Assertion failed: (isBitOrArrOfBits(rmap.second)), function runOnInstanceGraphNode, file verifyflattenedtypes.cpp, line 37.
Abort trap: 6
bash-3.2$ 
```

As you can see the error is the same when operating on json output of the flatten pass. Also, the flatten pass says the circuit is not modified during flattening. Am I doing something wrong here?

rdaly525, 2017-09-12:
Sorry this is an issue of a bad error message. What it means is that the type hierarchy has not been flattened. I assume that in verilog you first need to flatten the type hierarchy to be only bits or arrays of bits. 

But the main issue is that you do not need to (and should not run) the verilog pass directly. Specifying a ".v" extension will run the appropriate passes for you.

dillonhuff, 2017-09-12:
Awesome. That works.

Thanks!

closed 2017-09-12 by dillonhuff