rdaly525/coreir #198: Signedness issue with adding cout to add/sub

rdaly525, 2017-09-12:
Cout is calculated differently based on if the add/sub is signed or unsigned.

It seems like there are two options for the coreir ops. 

1. Change the single add op to be two ops: sadd, uadd (signed, unsigned). This is annoying because it is only relelvent to the cout bit.
2. Have two separate port options: unsigned_cout, signed_cout. You would be allowed to specify either of these ports. 

Thoughts?

leonardt, 2017-09-12:
Could we have a single COUT port and a parameter to the generator that specifies how COUT is calculated?

rdaly525, 2017-09-12:
I think that it what I was trying to say in #2. 
genargs for add: {"width" : int, "has_signed_cout": bool, "has_unsigned_cout": bool, "has_cin": bool}

leonardt, 2017-09-12:
I think I would vote that case, it seems simpler than splitting it into two primitives. Also, if has_cout=False, then sadd and uadd would be the same which seems unnecessary.

phanrahan, 2017-09-12:
I think COUT is the same if you add signed or unsigned numbers. If you sign extend the two numbers, you will get either 2 0's or 2 1's.  When these are added together you get 0, which is added to the carry out.

rdaly525, 2017-09-12:
Hmmm, I guess I am conflating "overflow" with "cout". 

Pat have you dealt with signed/unsigned overflow/underflow? This was in response to discussion about saturating math with the CGRA team. People want to add in direct support for overflow/underflow detection in each PE (instead of just a cout)

I guess I could add a non-primitive "add" module that could do overflow/underflow.

phanrahan, 2017-09-12:
Overflow is indeed different. That's a good point. I need to generate the overflow flag for my adders in order to get signed comparisons working in mantle. Overflow is typically denoted by V.

How about Add(n, cin, cout, vout). 

All combinations are useful!

I can send you the definitions ...

rdaly525, 2017-09-13:
Well since coreir primitives do not have the notion of signed and unsigned types like magma, I will need to have my interface be:

Add(width,has_cin,has_cout,has_signed_vout, has_unsigned_vout), or something similar.

phanrahan, 2017-09-13:
there is normally just a single vout. 

Remember, the addition circuit is exactly the same for signed and unsigned addition, including the calculation of cout. vout is an optional output flag that assumes the inputs are signed (since it is checking the sign bit (n). But vout can be computed for any numbers. On a CPU, there is a single add instruction that works for signed and unsigned numbers, and sets both the cout and vout flags. There is another instruction (adc) that includes the cin. Again, adc works the same for both signed and unsigned numbers, and normally sets cout and vout.

If you want to implement saturating arithmetic, then you need additional operators. And those depend on whether the inputs are signed or unsigned.

rdaly525, 2017-09-13:
I think I was not clear.

I am saying that I need to choose whether the vout port represents vout as if the inputs were signed or vout as if the inputs were unsigned. To be clear here are some examples;

```
//This will create an Add primitive where the vout is calculated assuming in0 and in1 are signed. 
Add(width=16,has_signed_vout=true)->getType() == 
  {"in0" : BitIn[16], 
    "in1" : BitIn[16],
    "out" : BitOut[16],
    "vout" : BitOut
} 

//This will create an Add primitive where the vout is calculated assuming in0 and in1 are unsigned signed. Also has a cin port

Add(width=16,has_cin=true,has_unsigned_vout=true)->getType() == 
  {"in0" : BitIn[16], 
    "in1" : BitIn[16],
    "cin" : BitIn,
    "out" : BitOut[16],
    "vout" : BitOut
  } 
```

This is similar to how I have to have two operations for "less than" (slt, ult).

If I just give it one option for a vout port, it is ambiguous whether it represents the result of a signed or unsigned overflow.

phanrahan, 2017-09-13:
They way I suggested is like a conventional ISA. Overflow is meant to indicate signed overflow, even though it is well defined for both signed and unsigned numbers. 

cout is the same for both signed and unsigned addition. You use cout when doing multiple precision arithmetic between both signed and unsigned numbers. This is the beauty of two's complement.

vout is only used for comparisons.

rdaly525, 2017-09-13:
Okay I think I understand the confusion. I wanted a single port which represented "This add operation either overflowed or underflowed". In reality for signed integers, **vout** should be used to represent (operation either overflowed or underflowed) whereas for unsigned integers, **cout** should be used to represent (operation overflowed).

I am not sure what you mean that vout is well defined for unsigned integers though. I understand you could calculate it using the formula you provided, but it seems like it does not represent anything useful (for unsigned integers).

phanrahan, 2017-09-13:
I think we are on the same page. 

Only thing I would add is the various conditions depend on z, n, c, and v. Here is a table from the ARM ISA docs.

[image]

rdaly525, 2017-10-04:
This is resolved. Only have cout and cin for add.

closed 2017-10-04 by rdaly525