rdaly525/coreir #205: [dev] JSON output is no longer consistent

leonardt, 2017-09-14:
magma/mantle regression tests are no longer working because linux/mac produce different orderings in the JSON outputs (inconsistency in the gold files between locally and on travis)

For our test cases the difference manifests in the ordering of genargs and the ordering of record type fields

leonardt, 2017-09-14:
This would become a non-issue once we migrate the magma/mantle tests to use the coreir simulator.  For now I might try using the verilog pass and verilator to run the tests.

rdaly525, 2017-09-14:
Can you send me two files which should be the same but have a different diff?

rdaly525, 2017-09-14:
Let me know if it is fixed in dev

leonardt, 2017-09-14:
Looks like the ordering of genargs is now consistent, but record type fields are still different.

Here's the failing test on travis: https://travis-ci.org/phanrahan/magma/jobs/275533240 (the gold file was generated locally on my mac)

Gold files: https://github.com/phanrahan/magma/tree/enable-coreir-regressions-tests/tests/test_coreir/gold
Test script: https://github.com/phanrahan/magma/blob/enable-coreir-regressions-tests/tests/test_coreir/test_coreir.py

phanrahan, 2017-09-14:
Per our recent conversation, I vote for making arguments (and any dictionary including record types) have a well-defined ordering. Hopefully that would help with the JSON.

rdaly525, 2017-09-14:
The likely cause of this is that I have Records being organized by *insertion order*. I did this early on because it is sometimes nice to maintain the order of the ports that were specified by the user.

The easy fix is for me to just sort by key like the rest of the dictionaries. Alternatively, magma could also sort by insertion order.

leonardt, 2017-09-16:
Fixed for magma/mantle tests

closed 2017-09-16 by leonardt