rdaly525/coreir #220: Add ability to call python code (magma) for generators and typegens.

rdaly525, 2017-09-21:
here is a good example of how to pass values  c <-> python. https://stackoverflow.com/questions/3286448/calling-a-python-method-from-c-c-and-extracting-its-return-value 

This will probably involve inheriting from TypeGen and GeneratorDef respectively.

rdaly525, 2017-09-25:
As a concrete task, start by writing a python generator for commonlib.LinebufferMem in terms of coreir.mem.

rdaly525, 2017-10-24:
partially fixed by #273

rdaly525, 2018-02-12:
@leonardt, What is the state of this issue at the moment?

leonardt, 2018-02-12:
We can call `typegen` from Python, we can dynamically create and define modules using the magma coreir backend, all we need left is the ability to call a `generator` from Python. This should look pretty much the same as the `typegen` code except instead of receiving a coreir `Type` as the return value, it should receive a coreir `Module`. Basically, we need to add a way to specify a generator function that is from Python.

leonardt, 2018-02-27:
Minimal working examples of this can be found at
https://github.com/leonardt/pycoreir/commit/6073db4d0abb68b57dada196f57769defdfb904b
https://github.com/leonardt/pycoreir/commit/663dae07928aadbbbdbd11f782da826a73330460

Closing this now, we can open more specific issues to be tracked (probably over at pycoreir or magma)

closed 2018-02-27 by leonardt