rdaly525/coreir #236: Rename simulator binary

leonardt, 2017-09-29:
@dillonhuff @rdaly525 Can we rename the simulator binary to something like `coreir-simulator` so we can install into the system path like the coreir binary? I don't have any strong feelings about the name, but I think something a bit more specific then `simulator` would be good to avoid any name conflicts.

Another idea could be integrate it with the existing `coreir` binary so there's one frontend for all the tools, not sure if this is possible or a good idea.

rdaly525, 2018-02-12:
I would prefer not to have this in the same binary and to keep this as a separate utility. I do agree about changing the name and including it in coreir releases as a different binary

rdaly525, 2019-03-28:
Closing as name has changed to coreir-simulator and coreir-interpreter

closed 2019-03-28 by rdaly525