rdaly525/coreir #237: Assertion error in simulator for add with cin

leonardt, 2017-09-29:
@dillonhuff Looks like it's failing because `mux` is the only 3-input operation supported.

```
❯ ~/repos/coreir/bin/simulator -i test_coreir/build/test_add_cin_two.json
topo_order.size() = 5
numVertices(g)    = 5
Topological order
4
3
2
0
1
Assertion failed: (false), function printTernop, file sim.cpp, line 217.
zsh: abort      ~/repos/coreir/bin/simulator -i test_coreir/build/test_add_cin_two.json
```

test_add_cin_two.json:
```
{"top":"global.Add4_cin",
"namespaces":{
  "global":{
    "modules":{
      "Add4_cin":{
        "type":["Record",{
          "I0":["Array",4,"BitIn"],
          "I1":["Array",4,"BitIn"],
          "O":["Array",4,"Bit"],
          "CIN":"BitIn"
        }],
        "instances":{
          "inst0":{
            "genref":"coreir.add",
            "genargs":{"has_cin":true, "has_cout":false, "width":4}
          }
        },
        "connections":[
          ["self.O","inst0.out"],
          ["self.I1","inst0.in1"],
          ["self.I0","inst0.in0"],
          ["self.CIN","inst0.cin"]
        ]
      }
    }
  }
}
}
```

dillonhuff, 2017-09-29:
Thanks for letting me know. I'm working on a fix.

From what I see in coreirprims.hpp only add and sub need cin and cout pins. Do other binary operators (like multiplication) need overflow bits as well?

rdaly525, 2017-10-03:
I think I merged this in. @leonardt, can you confirm that this bug can be closed?

leonardt, 2017-10-03:
Fixed

closed 2017-10-03 by leonardt