rdaly525/coreir #240: Simulator segfaulting on "configs" branch

rdaly525, 2017-10-04:
@dillonhuff, Can you take a look at this?

These were most of the changes I made to the API
```
s/Const(/Const::make(c,/
s/Args /Values /
s/AINT/c->Int()/
s/ABOOL/c->Bool()/
s/ASTRING/c->String()/
s/Config/Mod/
```

I also made each register initialization value to be a dynamic_bit_vector (which I typedef as BitVector)
I suspect this last one is causing the segfault? 

I also notice that when I print out the BitVector values, it seems i have garbage data.

dillonhuff, 2017-10-04:
I will take a look.

I do not understand your description of the changes to the api, for example what does:

```cpp
s/Const(/Const::make(c,/
```

mean?

rdaly525, 2017-10-04:
Sorry for not being clear. These were my vim search and replace commands while I was updating the repo haha

For example:

```
Context* c;
Args args({{"width",Const(16)}};
```

Would turn into:
```
Context* c;
Values args({{"width",Const::make(c,16)}};
```

rdaly525, 2017-10-04:
Feel free to stop by my office for any questions about the changes I made.

dillonhuff, 2017-10-04:
Yeah I will walk over there now. I just pulled the configs branch and it doesnt even build from scratch when I try it on my machine.

closed 2017-10-23 by rdaly525