rdaly525/coreir #266: Generated modules are never inlined?

dillonhuff, 2017-10-23:
I pulled the recent, large changes to dev and fixed most of the problems in the simulator. One problem: it seems that the flatten pass has changed so that instances that were generated from a generator are never inlined.

Specifically it looks like function ```inlineInstance``` always returns false without modifying the graph when ```isGen()``` is True, and ```isGen()``` returns true if the generator reference for the module is not null.

The full text of ```inlineInstance``` in my branch off of dev looks like so:

```cpp
//This will modify the moduledef to inline the instance
bool inlineInstance(Instance* inst) {
  //Special case for a passthrough
  //TODO should have a better check for passthrough than string compare
  if (inst->isGen() && inst->getGeneratorRef()->getName() == "passthrough") {
    inlinePassthrough(inst);
    return true;
  }
  if (inst->isGen()) {
    return false;
  }
  ModuleDef* def = inst->getContainer();
  Module* modInline = inst->getModuleRef();
  assert(modInline);

  if (!modInline->hasDef()) {
    cout << "Inline Pass: " << modInline->getName() << " has no definition, skipping..." << endl;
    return false;
  }
  
  //I will be inlining defInline into def
  //Making a copy because i want to modify it first without modifying all of the other instnaces of modInline
  ModuleDef* defInline = modInline->getDef()->copy();

  //Add a passthrough Module to quarentine 'self'
  addPassthrough(defInline->getInterface(),"_insidePT");
  

  string inlinePrefix = inst->getInstname() + "$";

  //First add all the instances of defInline into def with a new name
  for (auto instmap : defInline->getInstances()) {
    string iname = inlinePrefix + instmap.first;
    def->addInstance(instmap.second,iname);
  }
  
  //Now add all the easy connections (that do not touch the boundary)
  for (auto cons : defInline->getConnections()) {
    SelectPath pA = cons.first->getSelectPath();
    SelectPath pB = cons.second->getSelectPath();
    
    //Easy case: when neither are connect to self
    if (pA[0] != "self" && pB[0] != "self") {
      //Create the correct names and connect
      pA[0] = inlinePrefix + pA[0];
      pB[0] = inlinePrefix + pB[0];
      def->connect(pA,pB);
    }
  }
  
  //Create t3e Passthrough to quarentene the instance itself
  Instance* outsidePT = addPassthrough(inst,"_outsidePT");

  //Connect the two passthrough buffers together ('in' ports are facing the boundary)
  def->connect("_outsidePT.in",inlinePrefix + "_insidePT.in");

  //Now remove the instance (which will remove all the previous connections)
  def->removeInstance(inst);
  
  //Now inline both of the passthroughs
  inlineInstance(outsidePT);
  
  inlineInstance(cast<Instance>(def->sel(inlinePrefix + "_insidePT")));

  //typecheck the module
  def->validate();
  return true;
}

```

rdaly525, 2017-10-23:
@dillonhuff I just pushed a fix. Can you verify this issue is resolved?

dillonhuff, 2017-10-23:
Fixed.

Thanks!

closed 2017-10-23 by dillonhuff