rdaly525/coreir #267: Enable mux is not fully wired in linebuffer mem

dillonhuff, 2017-10-23:
Json example where inlined code is not fully connected:

```json
{"top":"global.lbMem",
"namespaces":{
  "global":{
    "modules":{
      "lbMem":{
        "type":["Record",{
          "clk":["Named","coreir.clkIn"],
          "wdata":["Array",2,"BitIn"],
          "rdata":["Array",2,"Bit"],
          "wen":"BitIn",
          "valid":"Bit"
        }],
        "instances":{
          "m0":{
            "genref":"commonlib.LinebufferMem",
            "genargs":{"depth":["Int",4], "width":["Int",2]}
          }
        },
        "connections":[
          ["m0.clk","self.clk"],
          ["m0.rdata","self.rdata"],
          ["m0.valid","self.valid"],
          ["m0.wdata","self.wdata"],
          ["m0.wen","self.wen"]
        ]
      }
    }
  }
}
}
```

rdaly525, 2017-10-24:
@dillonhuff, this can be closed right?

dillonhuff, 2017-10-24:
Yes.

closed 2017-10-24 by dillonhuff