rdaly525/coreir #282: Automatically discover g++-4.9 on linux

leonardt, 2017-10-26:
It seems like this is a standard binary name for g++-4.9 that can we try to find on linux if the default g++ version is not compatible.

rdaly525, 2018-02-12:
@leonardt was this resolved? Or is this needed?

leonardt, 2018-02-12:
Not resolved, still need to specify `export COREIRCONFIG="g++-4.9"`to compile on a standard ubuntu flavor (see Travis script). It's not **needed**, but definitely a quality of life improvement for linux users (don't have to pollute their environment with another variable).

You can close it if you don't want to support it, or wait for some linux user to come make the change (I haven't gotten to it since it doesn't affect me directly now that the travis script is stable).