rdaly525/coreir #284: Error when trying to produce verilog from conv_3_1

dillonhuff, 2017-10-27:
I've gotten the simulator to produce useable output for conv_3_1 from CGRAMapper/examples and now I am trying to generate verilog code to compare it to verilator.

When I try to emit verilog for the original file without running any passes the process works:

```bash
bash-3.2$ ./bin/coreir -i ~/CppWorkspace/CGRAMapper/examples/conv_3_1.json -o ./conv_3_1.v --load_libs ~/CppWorkspace/coreir/lib/libcoreir-commonlib.dylib
In Run Generators
g: commonlib.Linebuffer(bitwidth:16, image_width:10, stencil_height:3, stencil_width:1)
g: commonlib.LinebufferMem(depth:10, width:16)
g: coreir.add(width:16)
g: coreir.add(width:4)
g: coreir.const(width:16)
g: coreir.const(width:4)
g: coreir.mem(depth:10, width:16)
g: coreir.mul(width:16)
g: coreir.neq(width:4)
g: mantle.reg(has_clr:False, has_en:True, has_rst:False, width:4)
g: mantle.wire(type:Bit[16])
g: commonlib.Linebuffer(bitwidth:16, image_width:10, stencil_height:3, stencil_width:1)
g: commonlib.LinebufferMem(depth:10, width:16)
g: coreir.add(width:16)
g: coreir.add(width:4)
g: coreir.const(width:16)
g: coreir.const(width:4)
g: coreir.mem(depth:10, width:16)
g: coreir.mul(width:16)
g: coreir.mux(width:4)
g: coreir.neq(width:4)
g: coreir.reg(width:4)
g: mantle.reg(has_clr:False, has_en:True, has_rst:False, width:4)
g: mantle.wire(type:Bit[16])

Modified?: Yes

```

But when I try to first flatten and then wire up clocks so that it is like the coreir processed by the simulator I get an error in vmodule:

```bash
bash-3.2$ ./bin/coreir -i ~/CppWorkspace/CGRAMapper/examples/conv_3_1.json -o ./conv_3_1.v -p rungenerators,flattentypes,flatten,liftclockports-coreir,wireclocks-coreir --load_libs ~/CppWorkspace/coreir/lib/libcoreir-commonlib.dylib
In Run Generators
g: commonlib.Linebuffer(bitwidth:16, image_width:10, stencil_height:3, stencil_width:1)
g: commonlib.LinebufferMem(depth:10, width:16)
g: coreir.add(width:16)
g: coreir.add(width:4)
g: coreir.const(width:16)
g: coreir.const(width:4)
g: coreir.mem(depth:10, width:16)
g: coreir.mul(width:16)
g: coreir.neq(width:4)
g: mantle.reg(has_clr:False, has_en:True, has_rst:False, width:4)
g: mantle.wire(type:Bit[16])
g: commonlib.Linebuffer(bitwidth:16, image_width:10, stencil_height:3, stencil_width:1)
g: commonlib.LinebufferMem(depth:10, width:16)
g: coreir.add(width:16)
g: coreir.add(width:4)
g: coreir.const(width:16)
g: coreir.const(width:4)
g: coreir.mem(depth:10, width:16)
g: coreir.mul(width:16)
g: coreir.mux(width:4)
g: coreir.neq(width:4)
g: coreir.reg(width:4)
g: mantle.reg(has_clr:False, has_en:True, has_rst:False, width:4)
g: mantle.wire(type:Bit[16])
Inline Pass: neq has no definition, skipping...
Inline Pass: const has no definition, skipping...
Inline Pass: const has no definition, skipping...
Inline Pass: term has no definition, skipping...
Inline Pass: term has no definition, skipping...
Inline Pass: add has no definition, skipping...
Inline Pass: add has no definition, skipping...
Inline Pass: add has no definition, skipping...
Inline Pass: add has no definition, skipping...
Inline Pass: add has no definition, skipping...
Inline Pass: const has no definition, skipping...
Inline Pass: const has no definition, skipping...
Inline Pass: const has no definition, skipping...
Inline Pass: const has no definition, skipping...
Inline Pass: mem has no definition, skipping...
Inline Pass: mul has no definition, skipping...
Inline Pass: mul has no definition, skipping...
Inline Pass: mul has no definition, skipping...
Inline Pass: mux has no definition, skipping...
Inline Pass: reg has no definition, skipping...
In Run Generators
g: commonlib.Linebuffer(bitwidth:16, image_width:10, stencil_height:3, stencil_width:1)
g: commonlib.LinebufferMem(depth:10, width:16)
g: coreir.add(width:16)
g: coreir.add(width:4)
g: coreir.const(width:16)
g: coreir.const(width:4)
g: coreir.mem(depth:10, width:16)
g: coreir.mul(width:16)
g: coreir.mux(width:4)
g: coreir.neq(width:4)
g: coreir.reg(width:4)
g: mantle.reg(has_clr:False, has_en:True, has_rst:False, width:4)
g: mantle.wire(type:Bit[16])
Assertion failed: (0), function toConstString, file ../../../include/coreir/passes/analysis/vmodule.h, line 165.
Abort trap: 6

```

rdaly525, 2018-02-13:
@dillonhuff, is this still a bug?

rdaly525, 2018-03-16:
Just checked and I can produce verilog for this example.

closed 2018-03-16 by rdaly525