rdaly525/coreir #285: Error in fileReader.cpp when trying to read json for flattened LinebufferMem

dillonhuff, 2017-10-27:
In the unit tests for the interpreter (./tests/simulator/interpreter.cpp) I construct a LinebufferMem and save it to json:

```cpp
      uint index = 2;
      uint width = index;
      uint depth = pow(2, index);

      CoreIRLoadLibrary_commonlib(c);

      Type* lineBufferMemType = c->Record({
          {"clk", c->Named("coreir.clkIn")},
            {"wdata", c->BitIn()->Arr(width)},
              {"rdata", c->Bit()->Arr(width)},
        	{"wen", c->BitIn()},
        	  {"valid", c->Bit()}
        });

      Module* lbMem = c->getGlobal()->newModuleDecl("lbMem", lineBufferMemType);
      ModuleDef* def = lbMem->newModuleDef();

      def->addInstance("m0",
        	       "commonlib.LinebufferMem",
        	       {{"width", Const::make(c, width)},
        		   {"depth", Const::make(c, depth)}});

      def->connect("self.clk", "m0.clk");
      def->connect("self.wen", "m0.wen");
      def->connect("self.wdata", "m0.wdata");
      def->connect("m0.rdata", "self.rdata");
      def->connect("m0.valid", "self.valid");

      lbMem->setDef(def);

      c->runPasses({"rungenerators","flattentypes", "flatten"});

      if (!saveToFile(g, "linebuffermem.json",lbMem)) {
        cout << "Could not save to json!!" << endl;
        c->die();
      }

```

The resulting file is:

```json
{"top":"global.lbMem",
"namespaces":{
  "global":{
    "modules":{
      "lbMem":{
        "type":["Record",{
          "clk":["Named","coreir.clkIn"],
          "wdata":["Array",2,"BitIn"],
          "rdata":["Array",2,"Bit"],
          "wen":"BitIn",
          "valid":"Bit"
        }],
        "instances":{
          "m0$add_r":{
            "genref":"coreir.add",
            "genargs":{"width":["Int",2]}
          },
          "m0$add_w":{
            "genref":"coreir.add",
            "genargs":{"width":["Int",2]}
          },
          "m0$c1":{
            "genref":"coreir.const",
            "genargs":{"width":["Int",2]},
            "modargs":{"value":[["BitVector",2],1]}
          },
          "m0$mem":{
            "genref":"coreir.mem",
            "genargs":{"depth":["Int",4], "width":["Int",2]}
          },
          "m0$raddr$enMux":{
            "genref":"coreir.mux",
            "genargs":{"width":["Int",2]}
          },
          "m0$raddr$reg0":{
            "genref":"coreir.reg",
            "genargs":{"width":["Int",2]},
            "modargs":{"init":[["BitVector",2],"Arg","init"]}
          },
          "m0$veq":{
            "genref":"coreir.neq",
            "genargs":{"width":["Int",2]}
          },
          "m0$waddr$enMux":{
            "genref":"coreir.mux",
            "genargs":{"width":["Int",2]}
          },
          "m0$waddr$reg0":{
            "genref":"coreir.reg",
            "genargs":{"width":["Int",2]},
            "modargs":{"init":[["BitVector",2],"Arg","init"]}
          }
        },
        "connections":[
          ["m0$add_r.in0","m0$raddr$reg0.out"],
          ["m0$add_r.in1","m0$c1.out"],
          ["m0$add_r.out","m0$raddr$enMux.in1"],
          ["m0$add_w.in0","m0$waddr$reg0.out"],
          ["m0$add_w.in1","m0$c1.out"],
          ["m0$add_w.out","m0$waddr$enMux.in1"],
          ["m0$mem.clk","self.clk"],
          ["m0$mem.raddr","m0$raddr$reg0.out"],
          ["m0$mem.rdata","self.rdata"],
          ["m0$mem.waddr","m0$waddr$reg0.out"],
          ["m0$mem.wdata","self.wdata"],
          ["m0$mem.wen","self.wen"],
          ["m0$raddr$enMux.in0","m0$raddr$reg0.out"],
          ["m0$raddr$enMux.out","m0$raddr$reg0.in"],
          ["m0$raddr$enMux.sel","self.wen"],
          ["m0$raddr$reg0.clk","self.clk"],
          ["m0$raddr$reg0.out","m0$veq.in0"],
          ["m0$veq.in1","m0$waddr$reg0.out"],
          ["m0$veq.out","self.valid"],
          ["m0$waddr$enMux.in0","m0$waddr$reg0.out"],
          ["m0$waddr$enMux.out","m0$waddr$reg0.in"],
          ["m0$waddr$enMux.sel","self.wen"],
          ["m0$waddr$reg0.clk","self.clk"]
        ]
      }
    }
  }
}
}
```

When I try to use the coreir binary to write this file to json I get an error message:

```bash
bash-3.2$ ~/CppWorkspace/coreir/bin/coreir -i ./tests/simulator/linebuffermem.json -o ./linebuffermem.v
ERROR: NYI using Args here

Assertion failed: (m), function json2Value, file fileReader.cpp, line 321.
Abort trap: 6

```

Have you seen this message before? Am I doing something wrong here?

rdaly525, 2017-10-27:
You are not doing anything wrong, I just have not implemented this case. I am going to file a separate issue and will try to fix what I think is the bug.

dillonhuff, 2017-10-27:
Cool. Thanks!

closed 2017-10-28 by rdaly525