rdaly525/coreir #344: Inlining is slow for larger designs

dillonhuff, 2018-01-03:
Inlining an [old version of the 8 x 8 CGRA](https://github.com/dillonhuff/VerilogToCoreIR/tree/master/test/samples/genesis_verif) takes several minutes even though the design is only ~30,000 lines of JSON.

I did some preliminary profiling and it looks like the issue is that inlining passthroughs for complicated interfaces (like the PE tile interface) takes a very long time.

rdaly525, 2018-01-04:
Can you give me instructions on how to run this?

I am guessing I can run `coreir -p flatten` on some json?

dillonhuff, 2018-01-04:
@rdaly525 I just added the cgra json to the VerilogToCoreIR repo. The file is [here](https://github.com/dillonhuff/VerilogToCoreIR/blob/master/cgra_top_grouped.json)

run flattening on that file with:

```bash
bash-3.2$ coreir -i cgra_top_grouped.json -o cgra_run_gen.json -p rungenerators,flatten --load_libs /Users/dillon/CppWorkspace/coreir/lib/libcoreir-rtlil.dylib 
```

dillonhuff, 2018-01-04:
If you want to generate the json from scratch you should be able to do something like:

```bash
git clone https://github.com/dillonhuff/VerilogToCoreIR
cd VerilogToCoreIR
make -j
```

This will build the .so for the generate coreir pass and then run that pass on the samples in the ```test/samples``` directory. The json for the CGRA will be in top.json

rdaly525, 2018-01-05:
I am getting an error:

ERROR: MISSING rtlil.to_bv

It seems to be commented out in the rtlil lib in dev. Could you provide a fix?

dillonhuff, 2018-01-05:
Just added a fix for that and created a PR

rdaly525, 2018-01-05:
Turns out that the creating of the json symtables was the reason for it being super slow.

rdaly525, 2018-01-11:
Closing this as there exists a workaround

closed 2018-01-11 by rdaly525