rdaly525/coreir #358: Add a script to run pycoreir/magma/mantle tests on dev->master merges

leonardt, 2018-01-17:


rdaly525, 2018-02-12:
@leonardt, Any update on this issue?

leonardt, 2018-02-12:
Haven't had time to do this, have quite a few things on higher priority right now

leonardt, 2018-03-16:
Here's a script I have to run all the tests

```bash
#!/bin/bash

cd coreir
git checkout dev
git pull
make install -j 8
make test
cd ..

cd pycoreir
git checkout dev
git pull
pytest
cd ..

cd magma;
git checkout coreir-dev
git pull
pytest
cd ..

cd mantle;
git checkout coreir-dev
git pull
pytest
cd ..
```

Here's a script I have to open PRs to track the coreir PR
```python
#!/usr/bin/env python
import github3
from github3 import login, GitHub

g = login(<username>, <password>)

coreir = g.repository("rdaly525", "coreir")
coreir_pr = [_ for _ in coreir.pull_requests(state="open", head="dev", base="master")][0]

for owner, repo, branch in [
    ("leonardt", "pycoreir", "dev"),
    ("phanrahan", "magma", "coreir-dev"),
    ("phanrahan", "mantle", "coreir-dev")
    ]:
    body = f"Tracking coreir PR {coreir_pr.html_url}"
    title = f"Tracking rdaly525 PR {coreir_pr.number}"
    print(f"Creating PR for {owner}/{repo}:{branch} with title=\"{title}\" body=\"{body}\"")
    try:
        result = g.repository(owner, repo).create_pull(title=title, base="master", head=branch, body=body)
        print(result)
    except github3.exceptions.UnprocessableEntity as e:
        print(f"Caught exception: {e}")
        print("Could be that there are no changes between dev branch and master")
        print("FIXME: Should check that there are any changes to pull")
```

These could be improved, but is there a place I should actually check them in? They're at least documented in this issue now, but not sure where they would really belong

rdaly525, 2018-03-16:
This is great! Can you put it in tools/.

leonardt, 2018-03-22:
Resolved by https://github.com/rdaly525/coreir/pull/501

closed 2018-03-22 by leonardt