rdaly525/coreir #379: Pass "packbitconstants" assertion failure

makaimann, 2018-02-07:
I get the following complaint when trying to use the packbitconstants pass:
coreir: wiring_utils.cpp:185: CoreIR::Select* CoreIR::getDriverSelect(CoreIR::Select*): Assertion `false' failed.

In the attached file is a small example that just adds two inputs and produces that failure. I had to use .txt because GitHub didn't like .json but it's a coreir json file.

Thanks!

[adder.txt](https://github.com/rdaly525/coreir/files/1705189/adder.txt)

dillonhuff, 2018-02-08:
Unfortunately packbitconstants is currently a no-op, so there is no reason to use it anyway.

Do you actually need to pack bit constants into coreir constants? If so I can work on implementing it for real

dillonhuff, 2018-02-08:
Also for future reference could you send coreir json files with the .json extension just to keep things clean?

dillonhuff, 2018-02-08:
Ah, just read the second paragraph of your message. Thats very strange.

makaimann, 2018-02-08:
Yeah, I'm not sure why they only allow certain file extensions. 

And don't worry about it -- I don't actually need it. I just noticed the implementation of Mantle's Add with cin which uses a bunch of bit_gnd constants and thought it would be good to pack them. It's not necessary though.

rdaly525, 2018-02-12:
@dillonhuff is there a bug here?

dillonhuff, 2018-02-12:
No, I just need to implement this pass later if someone needs it

closed 2018-02-12 by dillonhuff