rdaly525/coreir #380: It is possible to create a connection from Bit[1] to Bit?

dillonhuff, 2018-02-08:
@rdaly525 for circuit specialization I'm testing some code that will disconnect a module port (say self.in) from all instances in a module and rewire the connections to a new constant value.

The replacement function [portToConstant](https://github.com/dillonhuff/coreir/blob/fc04c2c7518ed52c89af2503b0ec1ca98560ac31/src/simulator/wiring_utils.cpp#L323) always uses "coreir.const", with width equal to the port width, even when the port being replaced is a bit.

I figured I would need to write a special case that used corebit.const for bit types, so I wrote [this test case](https://github.com/dillonhuff/coreir/blob/fc04c2c7518ed52c89af2503b0ec1ca98560ac31/tests/unit/replace_port.cpp#L39) expecting a failure, but instead it passed.

The resulting json is:

```json
{"top":"global.port_in",
"namespaces":{
  "global":{
    "modules":{
      "port_in":{
        "type":["Record",{
          "in":"BitIn",
          "out":"Bit"
        }],
        "instances":{
          "def_self_const_replace_0":{
            "genref":"coreir.const",
            "genargs":{"width":["Int",1]},
            "modargs":{"value":[["BitVector",1],0]}
          },
          "neg":{
            "modref":"corebit.not"
          }
        },
        "connections":[
          ["def_self_const_replace_0.out","neg.in"],
          ["neg.out","self.out"]
        ]
      }
    }
  }
}
}
```

Is the connection: ```["def_self_const_replace_0.out","neg.in"]``` legal? I assumed it was not and that it would have to be: ```["def_self_const_replace_0.0.out","neg.in"]```

rdaly525, 2018-02-09:
That definitely should not be legal. Bit[1] != Bit Is it reporting that it works?

dillonhuff, 2018-02-09:
Yes. I can run that test case to completion and the result as a coreir json file

dillonhuff, 2018-02-09:
*and the json file I copied above is the result

rdaly525, 2018-02-12:
@dillonhuff, can you confirm this is resolved?

dillonhuff, 2018-02-12:
Yes, it is resolved.

closed 2018-02-12 by dillonhuff