rdaly525/coreir #421: Rework c-api memory model

leonardt, 2018-02-22:
For routines involving the creation of arrays (e.g. for passing a map as an array of keys and values), we should either:

- (a) have the user pass in the allocated array (so they'll need to call another API function to get the size of the array they should allocate). Then the user would be explicitly responsible for freeing the memory
- (b) define a calling convention such that it's the user's responsibility to free the memory (can be through an API call) after they're done with it

This should clean up the memory model such that the context doesn't need to managing memory for the c-api

rdaly525, 2018-02-23:
I like (b) a bit better. I think this is what LLVM does for their C api, although I will need to confirm

leonardt, 2018-02-23:
Makes sense to me