rdaly525/coreir #448: Simulator Should Clearly And Forcefully Complain When Improperly Setup

David-Durst, 2018-02-28:
I've had a lot of problems writing tests where I set the clock wrong or hadn't run the right passes. The error messages I got from the simulator were somewhat cryptic:

1. "Assertion failed: (inSels.size() == 2), function updateRegisterValue, file interpret.cpp, line 1360." when correct flatten passes hadn't run
2. "self.valid cannot be found" when I hadn't setup the simulator correctly by calling state.execute() at least once. 

Can the simulator add nice error messages for these user mistakes?

dillonhuff, 2018-03-06:
Error message 1. no longer exists due to a simulator refactor.

RE error message 2: In my opinion "self.valid cannot be found" is a clear and forceful complaint. That error originates in [a utility function](https://github.com/rdaly525/coreir/blob/d53a8c4f1a5a20b8ca9070ccafdde34dfab80f74/src/simulator/interpret.cpp#L566) that looks up select values. The function is used internally by the interpreter, and is available to the user in the interpreter API. It doesn't know why the select does not have a value, so it's error message can't tell the user what they need to do to fix the error.

The error message could offer a suggestion about possible sources of the error like clang or ghc sometimes do, but I cannot think of a concise, actionable suggestion to add to the warning that would be worth inserting and that covers all of the possible sources of the error.

rdaly525, 2018-03-06:
@dillonhuff, For Error message 1, I am guessing this is going to be in the next PR? Because I am still seeing it for some stuff I am doing in dev.

rdaly525, 2018-03-13:
@David-Durst, Can this be closed?

closed 2018-03-13 by David-Durst