rdaly525/coreir #449: VerifyConnectivity not catching unconnected port

David-Durst, 2018-02-28:
commonlib's serializer has a count port (https://github.com/David-Durst/coreir/blob/aetherlingMultipleParallelism/src/libs/commonlib.cpp#L1435)

I use serializer in my streamify (https://github.com/David-Durst/coreir/blob/aetherlingMultipleParallelism/src/libs/aetherlinglib/aeStreamifyArrayify.h#L56) but I never connect the count port to anything.

My streamify is used in a test that runs the verifyconnectivity pass (https://github.com/David-Durst/coreir/blob/aetherlingMultipleParallelism/tests/simulator/aetherlingHelpersSim.cpp#L134). 

Is this a bug or am I missing something?

rdaly525, 2018-02-28:
It looks like you are running verifyconnectivity-onlyinputs. count is an output port. So this is completely valid. If you want to verify the connectivity of both inputs and outputs, use verifyconnectivity

David-Durst, 2018-02-28:
I also use verifyconnectivity as the last pass.

rdaly525, 2018-03-16:
@David-Durst, was this resolved?