rdaly525/coreir #452: Change current simulator coreir.mem to have semantics of read before write instead of read after write

rdaly525, 2018-02-28:


dillonhuff, 2018-03-01:
Just to clarify: In [this test of simultaneous read and write to one memory location](https://github.com/dillonhuff/coreir/blob/85d22985d93f716b8bf53bb92c6cd05e2007085a/tests/simulator/interpreter.cpp#L1089-L1100) on [this line](https://github.com/dillonhuff/coreir/blob/85d22985d93f716b8bf53bb92c6cd05e2007085a/tests/simulator/interpreter.cpp#L1099) would you like the value of ```self.getBitVec("self.read_data")``` to be ```BitVec(width, 0)``` (its value before the last call to state.execute()) or ```BitVec(width, 23)``` ?

rdaly525, 2018-03-01:
I think it should be 23. You are setting the write value, then causing a rising edge of the clock, then checking the read value. If you checked the value of self.read_data before the execute statement, it should be 0.

dillonhuff, 2018-03-01:
Maybe we misunderstood each other yesterday. That is what the memory currently does.

I have added [some more tests](https://github.com/dillonhuff/coreir/blob/244adc8bfbad2dca3c3261bc23be7222eaed73d9/tests/simulator/interpreter.cpp#L1091-L1110) that show that memory read is combinational (changing raddr changes rdata on each execute call, regardless of whether a high clock edge arrived).

Could you send me a test case where the coreir.mem node is not doing what you would like?

closed 2018-03-08 by rdaly525