rdaly525/coreir #470: `coreir.neg` unsupported in interpreter

leonardt, 2018-03-09:
```
Unsupported node: neg_inst has operation name: coreir.neg
Assertion failed: (false), function updateNodeValues, file interpret.cpp, line 1160.
```

Steps to reproduce:
```
git clone -b test-coreir-neg https://github.com/leonardt/pycoreir
cd pycoreir
pytest -s
```

leonardt, 2018-03-09:
Also, `coreir.udiv`, `coreir.urem`, `coreir.sdiv`, `coreir.srem`, and `coreir.smod`

dillonhuff, 2018-03-15:
coreir.neg is fixed in #477

leonardt, 2018-03-15:
@dillonhuff Do you want me to open up a separate issue for `coreir.udiv`, `coreir.urem`, `coreir.sdiv`, `coreir.srem`, and `coreir.smod`? Or we can just keep this one open for them

dillonhuff, 2018-03-15:
@leonardt it would be good if you could open a separate issue for those.

Is implementation of them needed now?

leonardt, 2018-03-15:
@dillonhuff not needed now by me, they just failed in my unit tests for the primitives (since they weren't there), but I can wait to merge those tests in until the implementation is ready (and needed)

closed 2018-03-15 by leonardt