rdaly525/coreir #471: Assertion failure in interpreter, possibly nested arrays?

leonardt, 2018-03-12:
```
{"top":"global.Test",
"namespaces":{
  "global":{
    "modules":{
      "Test":{
        "type":["Record",[
          ["I",["Array",5,"BitIn"]],
          ["O",["Array",1,["Array",5,"Bit"]]]
        ]],
        "connections":[
          ["self.I","self.O.0"]
        ]
      }
    }
  }
}
}
```

Produces the following output when calling `state->execute();`
```
Starting topological sort
topo_order.size() = 2
numVertices(g)    = 2
Assertion failed: (bits.bitLength() == 1), function setValue, file interpret.cpp, line 1682.
```

Suspect nested arrays are the issue? @shacklettbp @dillonhuff any insight?

rdaly525, 2018-03-12:
I suspect that this is because the interpreter requires flattened types. You have to call the "flattentypes" pass in order to use the interpreter. 

@dillonhuff, perhaps always calling "verifyflattenedtypes" pass during initialization of the interpreter would help prevent these issues.

leonardt, 2018-03-12:
Makes sense. I can close this unless we plan to eventually support the ability to have non-flattened types? In that case, we can keep it open to track that feature.

rdaly525, 2018-03-14:
I am going to close this and move it to a new bug.

rdaly525, 2018-03-14:
Tracked in #476

closed 2018-03-14 by rdaly525