rdaly525/coreir #515: JSON serialization ordering no longer consistent

leonardt, 2018-04-16:
Seems to have been a regression in terms of output files being consistent between runs, I suspect from https://github.com/rdaly525/coreir/pull/514

To reproduce using pycoreir when on coreir's dev branch
```
# git clone -b coreir-dev git@github.com:leonardt/pycoreir.git
cd pycoreir
# pip install -e .
# pip install pytest
pytest
```
Running pytest multiple times will yield occasional failures when comparing the output `.json` files to gold files.

rdaly525, 2018-04-16:
Lenny, do you know if it is inconsistent with instance order, connections order, connection pair order, or all three?

leonardt, 2018-04-16:
Looks like connections in my tests

Output of 1 run
```
        "connections":[
          ["self.I.3","lut0.I3"],
          ["self.O","lut0.O"],
          ["self.I.0","lut0.I0"],
          ["self.I.1","lut0.I1"],
          ["self.I.2","lut0.I2"]
        ]
```

Output of another
```
        "connections":[
          ["self.I.0","lut0.I0"],
          ["self.I.1","lut0.I1"],
          ["self.I.2","lut0.I2"],
          ["self.I.3","lut0.I3"],
          ["self.O","lut0.O"]
        ]
```

leonardt, 2018-04-16:
In another case, it's connection pair order

Output of one run
```
        "connections":[
          ["self.in","mul.in0"],
          ["two.out","mul.in1"],
          ["mul.out","self.out"]
        ]
```

Output of another
```
        "connections":[
          ["mul.in0","self.in"],
          ["two.out","mul.in1"],
          ["mul.out","self.out"]
        ]
```

rdaly525, 2018-04-16:
Great, thanks I will try to fix today.

rdaly525, 2018-04-17:
@leonardt Are you still seeing an issue? Or can this be closed

leonardt, 2018-04-18:
Latest travis build for pycoreir fails https://travis-ci.org/leonardt/pycoreir/builds/367388218 after https://github.com/rdaly525/coreir/pull/517 was pulled. It looks like an issue with the ordering of the connections. I can reproduce it locally on `pycoreir` with the occasional failure (the two connections are occasionally swapped)

```
E                 -    ["self.in","mul.in0"],
E                 -           ["self.out","mul.out"],
E                 ? -------
E                 +    ["self.out","mul.out"],
E                 +           ["self.in","mul.in0"],...
```

leonardt, 2018-04-18:
Just re-ran the travis build I linked to make sure it's still happening

leonardt, 2018-04-19:
Fixed, thanks!

closed 2018-04-19 by leonardt