rdaly525/coreir #522: Simulator Issue (either bug or uninterpretable error message): inConns.size() == 1

David-Durst, 2018-04-21:
I get the below issue when running the test https://github.com/David-Durst/aetherling/blob/debuggingCoreIR/tests/test_partition.py with width / parallelism (lines 16 and 17) not equal to 2. The test passes if it is equal to 2. I believe all the connections are right because the graph is passing the verifyconnectivity-noclkrst pass and because I've hard errors with this message before. The flattened output is at https://github.com/David-Durst/aetherling/blob/debuggingCoreIR/tests/partition4_16.json.test. 

If this is my bug, how do I debug it? The simulator should give me a little more detail as to where the issue is.
```
Assertion failed: (inConns.size() == 1), function updateSliceNode, file interpret.cpp, line 538.
```

dillonhuff, 2018-04-22:
This should be fixed by #523

David-Durst, 2018-04-22:
Yes it does, thanks!

closed 2018-04-22 by David-Durst