rdaly525/coreir #524: Can't Find Item In Symbol Table.

David-Durst, 2018-04-23:
I get the below error when I do https://github.com/David-Durst/aetherling/blob/debuggingCoreIR/tests/test_map.py#L85 even though instance inst2$inst1$inst1 is clearly in the symbol table at https://github.com/David-Durst/aetherling/blob/debuggingCoreIR/tests/test_map_flattened.json.test (Note: the file is massive, I can't really get around in it in chrome, but you can clearly see that the file contains this in its symbol table if you search for it  "inst2$inst1$inst0.out_0":["inst2$inst1$inst1","in_0_data_0"]).

This is a problem with CoreIR, right? If its in the symbol table, shouldn't it be able to find the instance?

```
ERROR: Cannot find instance inst2$inst1$inst1

0   libcoreir.dylib                     0x000000010e6e5302 _ZN6CoreIR9ModuleDef3selERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEE + 818
1   libcoreir.dylib                     0x000000010e6e5559 _ZN6CoreIR9ModuleDef3selERKNSt3__15dequeINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEE + 185
2   libcoreir.dylib                     0x000000010e6e508f _ZN6CoreIR9ModuleDef3selERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEE + 191
3   libcoreir.dylib                     0x000000010ead5e35 _ZN6CoreIR14SimulatorState8setValueERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEERKN4bsim21quad_value_bit_vectorE + 53
4   libcoreir.dylib                     0x000000010ead5dd1 _ZN6CoreIR14SimulatorState8setValueERKNSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEERKN4bsim21quad_value_bit_vectorE + 65
5   libcoreir-c.dylib                   0x000000010e4b578f CORESimSetValue + 207
6   libffi.6.dylib                      0x000000010cba5884 ffi_call_unix64 + 76
7   ???                                 0x00007ffee44f0fa0 0x0 + 140732728807328
```

David-Durst, 2018-05-25:
@dillonhuff can you look at this?

dillonhuff, 2018-06-05:
The line in the testbench link is just a comment. Could you send an up to date link to the line where the crash happens?

dillonhuff, 2018-06-05:
Is the value you are setting during the crash a port of the top level module under test? Setting a wire value that is not an input is pointless (and therefor not allowed) because the set value will be erased on the next evaluate call anyway.

dillonhuff, 2018-06-05:
The behavior of the testbench itself was illegal. You cannot set internal wires in the simulator since that would make the internal state of the circuit inconsistent. I have added a more precise error message than the one that was there before.

closed 2018-06-08 by dillonhuff