rdaly525/coreir #527: Fold Constants pass produces incorrect values

makaimann, 2018-04-24:
This issue came up when using fold constants on a partially configured PE tile. Taking the and reduce of all zeros gets folded to the constant "true". I've shrunk the module down to be more readable, using only the modules which were incorrectly replaced. I haven't experimented with different combinations of andr/orr yet, so I don't know if it's the fact that they're connected that causes the problem or if this happens for any andr.

[fold-constants-bug.tar.gz](https://github.com/rdaly525/coreir/files/1940567/fold-constants-bug.tar.gz) contains the coreir json before and after running the fold constants pass.

makaimann, 2018-04-24:
Oh you know what, I didn't see the issue the first time I looked at the fold-constants pass, but on a second look I found it. It's a simple fix. I filed a pull request: https://github.com/rdaly525/coreir/pull/528

dillonhuff, 2018-04-24:
Good catch!

So do we still need to meet tomorrow or is this resolved?

makaimann, 2018-04-24:
Nope, we don't need to meet. It's resolved. Thanks though!

closed 2018-04-24 by makaimann