rdaly525/coreir #543: Generators Not Running And CLI Not Loading Dylib

David-Durst, 2018-05-21:
When I run the test at https://github.com/David-Durst/aetherling/blob/master/tests/test_serializer.py#L12-L40, the test fails because the generators aren't being run and then verifyconnectivity-noclkrst on line 40 doesn't know how to handle the generators. I have uploaded the files pre and post generators to show this. The are test_serializer_*.txt (renamed from .json for uploading here). Also, see below for why I can't run it from the command line:
```
(aetherling) durst@DN0a2302fb:~/dev/W17-8/aetherling$ pytest -s tests/test_serializer.py
============================================================================================================ test session starts =============================================================================================================
platform darwin -- Python 3.6.4, pytest-3.4.2, py-1.5.2, pluggy-0.6.0
rootdir: /Users/durst/dev/W17-8/aetherling, inifile:
collected 1 item

tests/test_serializer.py In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators
In Run Generators
Done running generators

running generators
In Run Generators
Done running generators
generators didn't fail
running wireclocks
wireclocks didn't fail
running verifyconnectivity-noclkrst
  Def:
    Instances:
      inst0 : serialize_tArray_11-In_Bit___n13
    Connections:
      inst0.I <=> self.I
      inst0.CLK <=> self.CLK
      inst0.out <=> self.O

ERROR: This is not a generated module!

0   libcoreir-c.dylib                   0x0000000106bed410 _ZN6CoreIR6Module10getGenArgsEv + 96
1   libcoreir.dylib                     0x0000000107104c9f _ZN6CoreIR6Passes18VerifyConnectivity21checkIfFullyConnectedEPNS_8WireableERNS_5ErrorE + 5647
2   libcoreir.dylib                     0x000000010710519b _ZN6CoreIR6Passes18VerifyConnectivity21checkIfFullyConnectedEPNS_8WireableERNS_5ErrorE + 6923
3   libcoreir.dylib                     0x0000000107106db8 _ZN6CoreIR6Passes18VerifyConnectivity11runOnModuleEPNS_6ModuleE + 936
4   libcoreir.dylib                     0x0000000106ebc069 _ZN6CoreIR11PassManager13runModulePassEPNS_4PassE + 1321
5   libcoreir.dylib                     0x0000000106ebe61a _ZN6CoreIR11PassManager7runPassEPNS_4PassE + 410
6   libcoreir.dylib                     0x0000000106ec1aee _ZN6CoreIR11PassManager3runENSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 5166
7   libcoreir.dylib                     0x0000000106ce06b6 _ZN6CoreIR7Context9runPassesENSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 150
8   libcoreir-c.dylib                   0x0000000106be8687 COREContextRunPasses + 2039
9   libffi.6.dylib                      0x00000001049b6884 ffi_call_unix64 + 76
10  ???                                 0x00007ffeebfd52c0 0x0 + 140732857668288
(aetherling) durst@DN0a2302fb:~/dev/W17-8/aetherling$
```

When I run the following command:
```
coreir -i test_serializer_nopasses.json -o test_serializer_cli_postrungenerators_postwireclocks.json -p 'rungenerators,wireclocks-coreir' -n 'aetherlinglib,commonlib,mantle,coreir,global' --load_libs '/usr/local/lib/libcoreir-aetherlinglib.dylib,/usr/local/lib/libcoreir-commonlib.dylib'
```
I get the below error (note that I can access aetherling.dehydrate in my other python tests, so I don't think the error is due to the compiled module missing the lib):
```
(aetherling) durst@DN0a2302fb:~/dev/W17-8/aetherling$ coreir -i test_serializer_nopasses.json -o test_serializer_cli_postrungenerators_postwireclocks.json -p 'rungenerators,wireclocks-coreir' -n 'aetherlinglib,commonlib,mantle,coreir,global' --load_libs '/usr/local/lib/libcoreir-aetherlinglib.dylib,/usr/local/lib/libcoreir-commonlib.dylib'
ERROR: Missing module: aetherlinglib.dehydrate

0   libcoreir.dylib                     0x000000010127fb24 _ZN6CoreIR7Context9getModuleENSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEE + 1060
1   libcoreir.dylib                     0x0000000101357119 _ZN6CoreIR10json2ValueEPNS_7ContextEN8nlohmann10basic_jsonINSt3__13mapENS4_6vectorENS4_12basic_stringIcNS4_11char_traitsIcEENS4_9allocatorIcEEEEbxydSA_NS2_14adl_serializerEEEPNS_6ModuleE + 3673
2   libcoreir.dylib                     0x00000001013533b5 _ZN6CoreIR11json2ValuesEPNS_7ContextEN8nlohmann10basic_jsonINSt3__13mapENS4_6vectorENS4_12basic_stringIcNS4_11char_traitsIcEENS4_9allocatorIcEEEEbxydSA_NS2_14adl_serializerEEEPNS_6ModuleE + 997
3   libcoreir.dylib                     0x000000010134c307 _ZN6CoreIR12loadFromFileEPNS_7ContextENSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEEPPNS_6ModuleE + 61847
4   coreir                              0x00000001010f4027 main + 44951
5   libdyld.dylib                       0x00007fff68c84015 start + 1
6   ???                                 0x000000000000000b 0x0 + 11
(aetherling) durst@DN0a2302fb:~/dev/W17-8/aetherling$
```

[test_serializer_nopasses.txt](https://github.com/rdaly525/coreir/files/2024149/test_serializer_nopasses.txt)
[test_serializer_postrungenerators_postwireclocks.txt](https://github.com/rdaly525/coreir/files/2024150/test_serializer_postrungenerators_postwireclocks.txt)

rdaly525, 2018-06-18:
@David-Durst, was this fixed?

David-Durst, 2018-07-13:
Fixed

closed 2018-07-13 by David-Durst