rdaly525/coreir #548: coreir distribution crashes when I try to use it?

dillonhuff, 2018-05-26:
@rdaly525 I've been experimenting with a JIT specializer for the CGRA in coreir.

I'm trying to benchmark it on kiwi and would like to get it building on travis so that I can debug linux compilation problems before running on kiwi (which is a pain to use via ssh from my mac).

I'm trying to download coreir as one of the dependencies. I can download the tarball, unpack it, and install coreir, but when I try to build the JIT on travis I get an error in the coreir json parser that my gcc version is not supported (even though I am setting COREIRCONFIG="g++4.9" in my travis file.

Any idea what might be wrong here?

[Link to travis build](https://travis-ci.org/dillonhuff/FlatCircuit)

leonardt, 2018-05-27:
I think the issue is that your `cmake` command is finding/using gcc 4.8. If we inspect the output of the cmake command we see
```
$ cmake .
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/travis/build/dillonhuff/FlatCircuit
```

One work around for this is if you want to point CMAKE to a specific compiler, you can try
```
cmake -DCMAKE_C_COMPILER=/usr/bin/gcc-4.9 -DCMAKE_CXX_COMPILER=/usr/bin/g++-4.9 .
```
or you can try
```
CC=gcc-4.9 CXX=gcc-4.9 cmake .
```

leonardt, 2018-05-27:
When compiling your project, it tries pulls in the `/usr/include/coreir/ir/json.h` header file which has a gcc version requirement

dillonhuff, 2018-05-27:
I tried the CMake change and it worked. Thanks!

closed 2018-05-27 by dillonhuff