rdaly525/coreir #567: Test Roms in simulator

rdaly525, 2018-06-30:
I changed the init parameter to roms to be a json argument. We should come up with multiple ways to interpret the rom parameter.
I am thinking something like:
{_type: "array", value: [[<double array of rom values>]]}
or
{_type: "file", value: "filename.bin"}

etc... This allows some flexibility in representing the initialization of ROMs.

rdaly525, 2018-06-30:
@dillonhuff , @leonardt What do you think?
We should also try to make something somewhat consistent with Magma

dillonhuff, 2018-07-11:
@rdaly525 now is not a great time since I'm preparing a paper for August 3rd.

Are there examples of coreir that use this new json rom? Also could you send a link to Pat's pico processor? Does it already have behavioral tests that I could use?