rdaly525/coreir #572: Building CoreIR on Linux

adamdai, 2018-07-11:
Hi, I'm trying to install CoreIR from source so I can use aetherling. I cloned the repo and switched to the dev branch, and running `make install CXX=g++-4.9` gives me the following error. 

```
make install CXX=g++-4.9
make -C src so
make[1]: Entering directory `/home/adam/coreir/src'
make -C simulator
make[2]: Entering directory `/home/adam/coreir/src/simulator'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/home/adam/coreir/src/simulator'
make -C passes
make[2]: Entering directory `/home/adam/coreir/src/passes'
make -C analysis
make[3]: Entering directory `/home/adam/coreir/src/passes/analysis'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/home/adam/coreir/src/passes/analysis'
make -C transform
make[3]: Entering directory `/home/adam/coreir/src/passes/transform'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/home/adam/coreir/src/passes/transform'
make[2]: Leaving directory `/home/adam/coreir/src/passes'
make -C ir so
make[2]: Entering directory `/home/adam/coreir/src/ir'
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/args.o args.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/common.o common.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/context.o context.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/coreirlib.o coreirlib.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/directedview.o directedview.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/dynamiclibrary.o dynamiclibrary.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/fileReader.o fileReader.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/fileWriter.o fileWriter.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/forcecast.o forcecast.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/generator.o generator.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/generatordef.o generatordef.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/globalvalue.o globalvalue.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/inline.o inline.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/instancegraph.o instancegraph.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/module.o module.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/moduledef.o moduledef.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/moduledef_validate.o moduledef_validate.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/namespace.o namespace.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/passes.o passes.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/passmanager.o passmanager.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/typecache.o typecache.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/typegen.o typegen.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/types.o types.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/valuecache.o valuecache.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/value.o value.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/valuetype.o valuetype.cpp
g++-4.9 -std=c++11 -Wall -fPIC -Werror -I../../include -I. -c -o build/wireable.o wireable.cpp
g++-4.9 -shared -o build/coreir.so build/args.o build/common.o build/context.o build/coreirlib.o build/directedview.o build/dynamiclibrary.o build/fileReader.o build/fileWriter.o build/forcecast.o build/generator.o build/generatordef.o build/globalvalue.o build/inline.o build/instancegraph.o build/module.o build/moduledef.o build/moduledef_validate.o build/namespace.o build/passes.o build/passmanager.o build/typecache.o build/typegen.o build/types.o build/valuecache.o build/value.o build/valuetype.o build/wireable.o ../passes/analysis/build/coreirjson.o ../passes/analysis/build/createcombview.o ../passes/analysis/build/createinstancegraph.o ../passes/analysis/build/createinstancemap.o ../passes/analysis/build/firrtl.o ../passes/analysis/build/hellomodule.o ../passes/analysis/build/magma.o ../passes/analysis/build/printer.o ../passes/analysis/build/smtlib2.o ../passes/analysis/build/smtmodule.o ../passes/analysis/build/smtoperators.o ../passes/analysis/build/smvmodule.o ../passes/analysis/build/smv.o ../passes/analysis/build/smvoperators.o ../passes/analysis/build/verifyconnectivity.o ../passes/analysis/build/verifyflatcoreirprims.o ../passes/analysis/build/verifyflattenedtypes.o ../passes/analysis/build/verifyinputconnections.o ../passes/analysis/build/verilog.o ../passes/analysis/build/vmodule.o ../passes/transform/build/adddirected.o ../passes/transform/build/add_dummy_inputs.o ../passes/transform/build/clockifyinterface.o ../passes/transform/build/cullgraph.o ../passes/transform/build/cullzexts.o ../passes/transform/build/deletedeadinstances.o ../passes/transform/build/delete_unused_inouts.o ../passes/transform/build/flatten.o ../passes/transform/build/flattentypes.o ../passes/transform/build/fold_constants.o ../passes/transform/build/packbitconstants.o ../passes/transform/build/packconnections.o ../passes/transform/build/registerinputs.o ../passes/transform/build/removebulkconnections.o ../passes/transform/build/removeconstduplicates.o ../passes/transform/build/removepassthroughs.o ../passes/transform/build/removeunconnected.o ../passes/transform/build/removewires.o ../passes/transform/build/rungenerators.o ../passes/transform/build/sanitize_names.o ../passes/transform/build/split_inouts.o ../passes/transform/build/transform2combview.o ../passes/transform/build/unpackconnections.o ../passes/transform/build/unresolvedsymbols.o ../passes/transform/build/wireclocks.o ../simulator/build/codegen.o ../simulator/build/dag_optimiziation.o ../simulator/build/interpret.o ../simulator/build/layout.o ../simulator/build/low_rep.o ../simulator/build/multithreading.o ../simulator/build/op_graph.o ../simulator/build/output.o ../simulator/build/print_c.o ../simulator/build/sim.o ../simulator/build/subcircuit.o ../simulator/build/utils.o ../simulator/build/wire_node.o ../simulator/build/wiring_utils.o -ldl
../simulator/build/dag_optimiziation.o: file not recognized: File truncated
collect2: error: ld returned 1 exit status
make[2]: *** [build/coreir.so] Error 1
rm build/dynamiclibrary.o build/instancegraph.o build/coreirlib.o build/generatordef.o build/inline.o build/context.o build/forcecast.o build/fileWriter.o build/valuecache.o build/passmanager.o build/typecache.o build/generator.o build/types.o build/typegen.o build/value.o build/module.o build/moduledef.o build/wireable.o build/passes.o build/common.o build/moduledef_validate.o build/valuetype.o build/directedview.o build/fileReader.o build/globalvalue.o build/args.o build/namespace.o
make[2]: Leaving directory `/home/adam/coreir/src/ir'
make[1]: *** [so] Error 2
make[1]: Leaving directory `/home/adam/coreir/src'
make: *** [build] Error 2
```
I also get the same error when running 'make -j'. Has anyone seen this error before or have any ideas how to get around it? Thanks

leonardt, 2018-07-11:
Was there an error in a previous run of the build? You may need to rebuild `dag_optimiziation.o`. Try deleting that file and re-running make

adamdai, 2018-07-11:
Oh alright thanks I think that worked. I wasn't able to run the tests because my VM ran out of virtual memory/swapspace but I'm guessing the install is fine...

closed 2018-07-11 by adamdai