rdaly525/coreir #573: Missing top module in verilog output

leonardt, 2018-07-12:
Input .json
```
{"top":"global.connect_box_width_width_16_num_tracks_10_has_constant1_default_value7_feedthrough_outputs_1111101111",
"namespaces":{
  "global":{
    "modules":{
      "DFF_init0_has_ceTrue_has_resetTrue":{
        "type":["Record",[
          ["I","BitIn"],
          ["O","Bit"],
          ["CLK",["Named","coreir.clkIn"]],
          ["CE","BitIn"],
          ["RESET","BitIn"]
        ]],
        "instances":{
          "bit_const_GND":{
            "modref":"corebit.const",
            "modargs":{"value":["Bool",false]}
          },
          "inst0":{
            "genref":"mantle.reg",
            "genargs":{"has_clr":["Bool",false], "has_en":["Bool",true], "has_rst":["Bool",false], "width":["Int",1]},
            "modargs":{"init":[["BitVector",1],"1'h0"]}
          },
          "inst1":{
            "modref":"global._Mux2"
          }
        },
        "connections":[
          ["inst1.I.1","bit_const_GND.out"],
          ["self.CLK","inst0.clk"],
          ["self.CE","inst0.en"],
          ["inst1.O","inst0.in.0"],
          ["self.O","inst0.out.0"],
          ["self.I","inst1.I.0"],
          ["self.RESET","inst1.S"]
        ]
      },
      "DFF_init1_has_ceTrue_has_resetTrue":{
        "type":["Record",[
          ["I","BitIn"],
          ["O","Bit"],
          ["CLK",["Named","coreir.clkIn"]],
          ["CE","BitIn"],
          ["RESET","BitIn"]
        ]],
        "instances":{
          "bit_const_VCC":{
            "modref":"corebit.const",
            "modargs":{"value":["Bool",true]}
          },
          "inst0":{
            "genref":"mantle.reg",
            "genargs":{"has_clr":["Bool",false], "has_en":["Bool",true], "has_rst":["Bool",false], "width":["Int",1]},
            "modargs":{"init":[["BitVector",1],"1'h1"]}
          },
          "inst1":{
            "modref":"global._Mux2"
          }
        },
        "connections":[
          ["inst1.I.1","bit_const_VCC.out"],
          ["self.CLK","inst0.clk"],
          ["self.CE","inst0.en"],
          ["inst1.O","inst0.in.0"],
          ["self.O","inst0.out.0"],
          ["self.I","inst1.I.0"],
          ["self.RESET","inst1.S"]
        ]
      },
[... 3228 lines cut ...]
  assign inst8__I[6] = I6[8];
  assign inst8__I[7] = I7[8];
  assign inst8__I[8] = I8[8];
  assign inst8__I[9] = I9[8];
  assign O[8] = inst8__O;
  assign inst9__I[0] = I0[9];
  assign inst9__I[10] = I10[9];
  assign inst9__I[11] = I11[9];
  assign inst9__I[12] = I12[9];
  assign inst9__I[13] = I13[9];
  assign inst9__I[14] = I14[9];
  assign inst9__I[15] = I15[9];
  assign inst9__I[1] = I1[9];
  assign inst9__I[2] = I2[9];
  assign inst9__I[3] = I3[9];
  assign inst9__I[4] = I4[9];
  assign inst9__I[5] = I5[9];
  assign inst9__I[6] = I6[9];
  assign inst9__I[7] = I7[9];
  assign inst9__I[8] = I8[9];
  assign inst9__I[9] = I9[9];
  assign O[9] = inst9__O;

endmodule //Mux16x16
```

leonardt, 2018-07-12:
Sorry, those files were too large to easily read the issue, i've attached them instead (needed to rename them to `.txt` so github would let me upload them into the issue).

leonardt, 2018-07-12:
Possibly related to https://github.com/rdaly525/coreir/issues/568?

leonardt, 2018-07-12:
Looks like this bug was introduced in the most recent dev->master merge.
Reverting to the previous release (`git checkout git checkout f9c60fa44a3ac6f5bee85f628988716786a0fa9c`) fixes the issue.

leonardt, 2018-07-13:
Sorry typo in the last comment, the working commit is dd081a2d6d1f8268cd4034fda88b28bd8ab33863

rdaly525, 2018-07-20:
Should be fixed by #582

leonardt, 2018-07-20:
Fixed by #582

leonardt, 2018-07-20:
@adai you can now update your coreir installation to the latest master branch

closed 2018-07-20 by leonardt