rdaly525/coreir #579: Generators Not Running (Again)

David-Durst, 2018-07-17:
This seems related to https://github.com/rdaly525/coreir/issues/543, though that particular file now works.

One of my Linebuffer tests fails at the line https://github.com/David-Durst/aetherling/blob/feab7925fae570fe345c80a50c21fd2c60a3fdc5/tests/test_linebuffer.py#L58. It appears to be because the generators aren't running, as with the prior issue. I get the same issue when running it through the CLI (see below). Thoughts?

The error message is:
```
(aetherling) durst@DN800caf06:~/dev/W17-8/aetherling$ pytest -s tests/test_linebuffer.py 
======================================================================================================= test session starts =======================================================================================================
platform darwin -- Python 3.6.4, pytest-3.4.2, py-1.5.2, pluggy-0.6.0
rootdir: /Users/durst/dev/W17-8/aetherling, inifile:
collected 3 items                                                                                                                                                                                                                 

tests/test_linebuffer.py In Run Generators
Done running generators
.In Run Generators
Done running generators
running generators
In Run Generators
Done running generators
generators didn't fail
running wireclocks
wireclocks didn't fail
running verifyconnectivity-noclkrst
  Def:
    Instances:
      bit_const_VCC : const
      inst0 : linebuffer(has_valid:True, image_type:BitIn[8][30], input_type:BitIn[8][2], is_last_lb:True, output_type:Bit[8][4])
    Connections:
      inst0.in <=> self.I
      bit_const_VCC.out <=> inst0.wen
      inst0.out <=> self.O
      inst0.valid <=> self.valid
      inst0.clk <=> self.CLK

ERROR: This is not a generated module!

0   libcoreir-c.dylib                   0x0000000110309e44 _ZN6CoreIR6Module10getGenArgsEv + 100
1   libcoreir.dylib                     0x000000011084653f _ZN6CoreIR6Passes18VerifyConnectivity21checkIfFullyConnectedEPNS_8WireableERNS_5ErrorE + 5647
2   libcoreir.dylib                     0x0000000110846a3b _ZN6CoreIR6Passes18VerifyConnectivity21checkIfFullyConnectedEPNS_8WireableERNS_5ErrorE + 6923
3   libcoreir.dylib                     0x0000000110848658 _ZN6CoreIR6Passes18VerifyConnectivity11runOnModuleEPNS_6ModuleE + 936
4   libcoreir.dylib                     0x00000001105e6d4e _ZN6CoreIR11PassManager13runModulePassEPNS_4PassE + 1326
5   libcoreir.dylib                     0x00000001105e93aa _ZN6CoreIR11PassManager7runPassEPNS_4PassE + 410
6   libcoreir.dylib                     0x00000001105ec8b6 _ZN6CoreIR11PassManager3runENSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 5190
7   libcoreir.dylib                     0x00000001104001a6 _ZN6CoreIR7Context9runPassesENSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 150
8   libcoreir-c.dylib                   0x0000000110305419 COREContextRunPasses + 2105
9   libffi.6.dylib                      0x000000010e089884 ffi_call_unix64 + 76
10  ???                                 0x00007ffee2902010 0x0 + 140732699516944
```

When I run the first three passes from the CLI though, it fails with the same error
 (see [test_linebuffer24.json.txt](https://github.com/rdaly525/coreir/files/2203041/test_linebuffer24.json.txt) for the input to this command)
```
(aetherling) durst@DN800caf06:~/dev/W17-8/aetherling$ coreir -i test_linebuffer24.json -o test_linebuffer24_cli_postrungenerators_postwireclocks.json -p 'rungenerators,wireclocks-coreir,verifyconnectivity-noclkrst' -n 'aetherlinglib,commonlib,mantle,coreir,global' --load_libs 'libcoreir-aetherlinglib.dylib,libcoreir-commonlib.dylib'
In Run Generators
Done running generators
  Def:
    Instances:
      bit_const_VCC : const
      inst0 : linebuffer(has_valid:True, image_type:BitIn[8][30], input_type:BitIn[8][2], is_last_lb:True, output_type:Bit[8][4])
    Connections:
      inst0.out <=> self.O
      inst0.valid <=> self.valid
      inst0.clk <=> self.CLK
      inst0.in <=> self.I
      bit_const_VCC.out <=> inst0.wen

ERROR: This is not a generated module!

0   libcoreir.dylib                     0x000000010c95a344 _ZN6CoreIR6Module10getGenArgsEv + 100
1   libcoreir.dylib                     0x000000010ce0e53f _ZN6CoreIR6Passes18VerifyConnectivity21checkIfFullyConnectedEPNS_8WireableERNS_5ErrorE + 5647
2   libcoreir.dylib                     0x000000010ce0ea3b _ZN6CoreIR6Passes18VerifyConnectivity21checkIfFullyConnectedEPNS_8WireableERNS_5ErrorE + 6923
3   libcoreir.dylib                     0x000000010ce10658 _ZN6CoreIR6Passes18VerifyConnectivity11runOnModuleEPNS_6ModuleE + 936
4   libcoreir.dylib                     0x000000010cbaed4e _ZN6CoreIR11PassManager13runModulePassEPNS_4PassE + 1326
5   libcoreir.dylib                     0x000000010cbb13aa _ZN6CoreIR11PassManager7runPassEPNS_4PassE + 410
6   libcoreir.dylib                     0x000000010cbb48b6 _ZN6CoreIR11PassManager3runENSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 5190
7   libcoreir.dylib                     0x000000010c9c81a6 _ZN6CoreIR7Context9runPassesENSt3__16vectorINS1_12basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEENS6_IS8_EEEESA_ + 150
8   coreir                              0x000000010c8487f4 main + 26420
9   libdyld.dylib                       0x00007fff72a2d015 start + 1
```

leonardt, 2018-07-17:
Any idea when this was introduced? You might try rolling back to a prior release (see this page for the commiits to checkout https://github.com/rdaly525/coreir/releases) to see when the bug was introduced

David-Durst, 2018-07-17:
Verified fails with same error for v0.0.13 and v0.0.10. v0.0.10 is from March 16. This is either a bug with my code or its been there for a long time.

rdaly525, 2018-07-20:
This was just an error in the error reporting. Really there is a bug in the implementation of the linebuffer (unconnected wire). Referenced in #580

David-Durst, 2018-07-20:
The test passes now that I wired valid_chain up to a term.

closed 2018-07-20 by David-Durst