rdaly525/coreir #587: Mapping magma's unnamed tuple type to coreir

leonardt, 2018-07-30:
Magma has a tuple type that is similar to Python's tuple, which is a heterogenous container (unlike array) that can be indexed. The python version:
```
a = (False, [2, 3], 32)
print(a[1])
```
I tried mapping the type to a coreir record with the index as the key ("0", "1"), but coreir complains that on the variable name with
```
ERROR: 0 0: is not a valid coreIR name!. Needs to be = ^[a-zA-Z_\-\$][a-zA-Z0-9_\-\$]*
```

Here's some possible resolutions:
* magma maps the index to a string key (e.g. "one", "two", ...)
* coreir supports string numbers or integers as record keys
* coreir adds a tuple type

@rdaly525 thoughts?

rdaly525, 2018-07-30:
I would recommend ("_0", "_1", "_2", etc) since that is easy to programmatically recognize and construct. The second option is possible, but there are some disadvantages since currently I can use numbers in select paths to disambiguate between records and arrays. I would rather not do the third option since it is simply a more restrictive version of a record type and can be easily emulated using a record type.

leonardt, 2018-07-30:
Okay, seem reasonable. I'll go with the "_" prefix for now, that was simple enough.

closed 2018-07-30 by leonardt