rdaly525/coreir #593: Failing CoreIR Clock Wiring Pass On Nested Map Call

David-Durst, 2018-08-05:
https://github.com/phanrahan/magma/issues/238 enabled the wire-clocks pass to handle modules with an input that is an array of CLK inputs. However, the pass still fails for modules with an input that is an array of arrays of CLK inputs. Running https://github.com/David-Durst/aetherling/blob/2b04d346dcd82dee3691c4e503b9859d1dbef010/tests/test_native_line_buffer.py#L58-L80 fails with the following output:

```Shell
Testing started at 11:34 PM ...
/Users/durst/anaconda/envs/aetherling/bin/python "/Users/durst/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/182.3684.100/PyCharm.app/Contents/helpers/pydev/pydevd.py" --multiproc --qt-support=auto --client 127.0.0.1 --port 64759 --file "/Users/durst/Library/Application Support/JetBrains/Toolbox/apps/PyCharm-P/ch-0/182.3684.100/PyCharm.app/Contents/helpers/pycharm/_jb_pytest_runner.py" --target test_native_line_buffer.py::test_double_nested_sipo -- -s
Connected to pydev debugger (build 182.3684.100)
pydev debugger: process 36570 is connecting

Launching pytest with arguments -s test_native_line_buffer.py::test_double_nested_sipo in /Users/durst/dev/W17-8/aetherling/tests

============================= test session starts ==============================
platform darwin -- Python 3.6.4, pytest-3.4.2, py-1.5.2, pluggy-0.6.0
rootdir: /Users/durst/dev/W17-8/aetherling, inifile:
collected 1 item
test_native_line_buffer.py In Run Generators
Done running generators
In Run Generators
Done running generators

Assertion failed: (0), function die, file context.cpp, line 112.
Running pass rungenerators
In Run Generators
Done running generators
Running pass wireclocks-coreir
ERROR: mapParallel: Cannot wire together
  self.CLK : coreir.clk[1][1]
  op_0.CLK[0] : coreir.clkIn

I AM DYING!

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

leonardt, 2018-08-06:
I don't know if we can automatically wire up an array of clock inputs.

For example, if we have
```
self.CLK : coreir.clk[2]
```
and we have two instances
```
inst0.CLK
inst1.CLK
```
which index of self.CLK should be wired up to inst0 or inst1?

leonardt, 2018-08-06:
If it's a 1 element array we could do it, but that seems like a funky special case

leonardt, 2018-08-06:
I think in this case, it might be best of the generator creating the instances and clock inputs handles the wiring itself? Because in the case of map I'm guessing there's a one to one mapping (clk[i] wires up to inst{i}.clk). Don't know if we can generalize this notion.

leonardt, 2018-08-06:
So you're asking for wireclocks to handle just a single source clock as an array of 1 element, possibly nested?

leonardt, 2018-08-06:
You're original issue asked for support for an array of of CLK inputs, which implies more than one clock input

leonardt, 2018-08-06:
Oh I see, you're just asking for a nested array of arrays of clock inputs to an instance, not in the interface, the wording is confusing.

leonardt, 2018-08-06:
I think there is still an ordering issue, see this module definition
```
Module: mapParallel(numInputs:1, operator:aetherlinglib.mapParallel)
  Type: {'O':Bit[4][1][1], 'CE':BitIn[1][1], 'CLK':coreir.clkIn[1][1], 'I':BitIn[1][1]}
  Def? Yes
  Def:
    Instances:
      op_0 : mapParallel(numInputs:1, operator:global.SIPO4CE)
    Connections:
      op_0.CE <=> self.CE[0]
      op_0.CLK <=> self.CLK[0][0]
      op_0.I <=> self.I[0]
      op_0.O <=> self.O[0]
```

In this case, we could automatically do the wiring `op_0.CLK <=> self.CLK[0][0]`, which I think is what you want, but that only works if `self.CLK` is a nested array of just 1 element. So this would be special casing wiring a single source clock that just happens to be described as a nested array. If we have a general nested array of arbitrary length, we can't wire that up automatically.

We can add this as a special case to the wire clocks pass

leonardt, 2018-08-07:
@David-Durst the test now passes for me with d9429b4abddd1acbd64b7739b277ef69417bc795, can you try it out? Changed the logic to detect if a clock is connected to be quite a bit simpler.

David-Durst, 2018-08-07:
@leonardt it works for me now in the doubly nested case. Thanks!

closed 2018-08-07 by David-Durst