rdaly525/coreir #621: Mechanism for compiling wrapped verilog (black box)

leonardt, 2018-09-26:
Is there an existing mechanism for attaching an existing verilog file or string to a module? That is, the module is a "declaration" (with an interface) and when compiled through the verilog backend, it just emits the attached string.

rsetaluri, 2018-09-26:
I started working on this and it is definitely possible, but didn't finish because I realized using `DeclareFromVerilog` in magma was good enough. The way to do it is to add metadata to the declaration and check in coreir if a `verilog` metadata item exists, then emit that as the generated verilog. There are some nuances in terms of whether to take the module declaration/header from the original source or use the codegen for it. I think we should ideally copy and paste the entire verilog module, not just the body. This is harder to do (afaik) with the current coreir codepaths.

rsetaluri, 2018-09-27:
@rdaly525 to update you: @leonardt and I had some discussions. I think ultimately it would be good to have explicit verilog wrapped modules in coreir. Whatever the specific mechanism is, the functionality we want is to declare a module in CoreIR with an interface specified in CoreIR, that when compiled to Verilog splats out a given piece of code **including** the header/declaration as well. That is, in generating Verilog code the **only** thing we do is copy and paste some string.

I believe most of this functionality is in there, except the ability to override the codegen for the declaration. @rdaly525 is this something we can add?

rdaly525, 2018-10-01:
@rsetaluri I am a little confused on what you want here. Lets discuss tomorrow.

rsetaluri, 2018-10-05:
Updating based on recent discussions:

1. We want a mechanism to c/p an *entire* verilog module, including interface. To enable this, we should add something like a `interface_and_definition` inside `module.metadata.verilog`. If this is present, then both `interface` and `definition` are ignored, and the entirety of `interface_and_definition` is used as the verilog codegen (let's pick a better name than `interface_and_definition`)

2. For parametrized modules, we need to do some work on the magma side to parse the parameter interface and then provide an abstract representation to CoreIR so it can formalize the type gens. Then CoreIR can still c/p the entire interface+definition of the verilog, but also has a representation to do type checking.

rsetaluri, 2018-10-18:
Here is a small spec for the verilog metadata:

```json
"verilog" = {
  "module?": <module>,
  "prefix": <prefix>,
  "definition": <definition>,
  "interface": <interface>,
  "parameters": [<param>, <param>, ...],
  "inlineable": <inlineable>
}
```

If `"module"` exists as a field then it is assumed that this is the entirety of the verilog module (interface and definition), and it should be copy pasted. In this case, the fields `"definition"` and `"interface"` should not exist.

closed 2018-11-02 by rsetaluri