rdaly525/coreir #622: Possible issue with clang/llvm 10.0.0 (latest xcode tools)

leonardt, 2018-10-02:
Coreir version
```
❯ git show
commit 8f2405acfd4483f863586e6bc849ce2073d7ade8 (HEAD -> master, tag: v0.0.26, origin/master, origin/HEAD)
Author: Ross <rdaly525@stanford.edu>
Date:   Tue Oct 2 15:12:53 2018 -0700
    bumped version
diff --git a/version.txt b/version.txt
index 337ad328..c55013bc 100644
--- a/version.txt
+++ b/version.txt
@@ -1 +1 @@
-v0.0.24
+v0.0.25
```
Result of trying to compile
```
❯ make install
/Library/Developer/CommandLineTools/usr/bin/make -C src dylib
/Library/Developer/CommandLineTools/usr/bin/make -C simulator
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/codegen.o codegen.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/dag_optimiziation.o dag_optimiziation.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/interpret.o interpret.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/layout.o layout.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/low_rep.o low_rep.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/multithreading.o multithreading.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/op_graph.o op_graph.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/output.o output.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/print_c.o print_c.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/sim.o sim.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/subcircuit.o subcircuit.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/utils.o utils.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/wire_node.o wire_node.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../..//include -I. -c -o build/wiring_utils.o wiring_utils.cpp
/Library/Developer/CommandLineTools/usr/bin/make -C passes
/Library/Developer/CommandLineTools/usr/bin/make -C analysis
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/coreirjson.o coreirjson.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/createcombview.o createcombview.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/createinstancegraph.o createinstancegraph.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/createinstancemap.o createinstancemap.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/firrtl.o firrtl.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/hellomodule.o hellomodule.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/magma.o magma.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/printer.o printer.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/smtlib2.o smtlib2.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/smtmodule.o smtmodule.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/smtoperators.o smtoperators.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/smv.o smv.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/smvmodule.o smvmodule.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/smvoperators.o smvoperators.cpp
g++ -std=c++11 -Wall -fPIC -Werror -I../../../include -I. -c -o build/verifyconnectivity.o verifyconnectivity.cpp
In file included from verifyconnectivity.cpp:1:
In file included from ../../../include/coreir.h:5:
In file included from ../../../include/coreir/ir/casting/casting.h:26:
/Library/Developer/CommandLineTools/usr/include/c++/v1/memory:3656:5: error: destructor called on non-final 'cxxopts::values::standard_value<bool>' that has virtual functions but non-virtual destructor [-Werror,-Wdelete-non-virtual-dtor]
    __data_.second().~_Tp();
    ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/memory:3612:5: note: in instantiation of member function 'std::__1::__shared_ptr_emplace<cxxopts::values::standard_value<bool>, std::__1::allocator<cxxopts::values::standard_value<bool> >
      >::__on_zero_shared' requested here
    __shared_ptr_emplace(_Alloc __a)
    ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/memory:4277:26: note: in instantiation of member function 'std::__1::__shared_ptr_emplace<cxxopts::values::standard_value<bool>, std::__1::allocator<cxxopts::values::standard_value<bool> >
      >::__shared_ptr_emplace' requested here
    ::new(__hold2.get()) _CntrlBlk(__a2, _VSTD::forward<_Args>(__args)...);
                         ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/memory:4656:29: note: in instantiation of function template specialization 'std::__1::shared_ptr<cxxopts::values::standard_value<bool> >::make_shared<>' requested here
    return shared_ptr<_Tp>::make_shared(_VSTD::forward<_Args>(__args)...);
                            ^
../../../include/coreir/tools/cxxopts.h:740:17: note: in instantiation of function template specialization 'std::__1::make_shared<cxxopts::values::standard_value<bool>>' requested here
    return std::make_shared<values::standard_value<T>>();
                ^
../../../include/coreir/tools/cxxopts.h:1008:22: note: in instantiation of function template specialization 'cxxopts::value<bool>' requested here
        = ::cxxopts::value<bool>(),
                     ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/memory:3656:23: note: qualify call to silence this warning
    __data_.second().~_Tp();
                      ^
1 error generated.
make[3]: *** [build/verifyconnectivity.o] Error 1
make[2]: *** [all] Error 2
make[1]: *** [dylib] Error 2
make: *** [build] Error 2
```

Here's my `g++` version
```
❯ g++ --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 10.0.0 (clang-1000.10.44.2)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

leonardt, 2018-10-02:
Here's a similar issue: https://github.com/aws/aws-sdk-cpp/issues/825
Here is there resolution: https://github.com/aws/aws-sdk-cpp/pull/828/files

leonardt, 2018-10-02:
Temporary workaround:
```
brew install gcc
make CXX=g++-8 install -j 8
```

leonardt, 2018-10-03:
Current workaround (depends on https://github.com/rdaly525/coreir/pull/624)
```
brew install gcc@-4.9
make CXX=g++-4.9
```

leonardt, 2018-10-03:
`g++-4.9` spits out a bunch of strange text segment bugs. Using `g++-8` (from `brew install gcc`) avoids these warnings.

rdaly525, 2018-10-30:
Fixed by updating cxxopts #636

leonardt, 2018-10-30:
Confirmed that #636 fixes it for me, no longer need to use a different g++ to compile.

closed 2018-10-30 by leonardt