rdaly525/coreir #625: Latest coreir release prepending `global_` to output verilog modules in the global namspace

leonardt, 2018-10-04:
Is this behavior intended? The main issue here is the top module in the `.json` file now has a different name in the output verilog, specifically `global_<top_name>`. We could handle this mapping downstream (i.e. in the magma/garnet code), but I wanted to double check that this is the intended behavior. It's a little odd that the `top` module that I pass in to coreir now has a different name in the output verilog, but it could be okay if we always assume that we just have to prepend `global_` to the name.

leonardt, 2018-10-04:
Here's an example 
[example.json.txt](https://github.com/rdaly525/coreir/files/2446767/example.json.txt)

```
❯ mv example.json.txt example.json  # rename github upload
❯ cat example.json | grep top  # top module
{"top":"global.PE__ops_add_sub_and__or___UIntKind_16",
❯ coreir -i example.json -o example.v
❯ cat example.v | grep PE__ops_add_sub_and__or___UIntKind_16  # top has global_ prepended
module global_PE__ops_add_sub_and__or___UIntKind_16 (
```

rdaly525, 2018-10-10:
Fixed by #628

closed 2018-10-10 by rdaly525